ADTPro/adtpro #93: Raspberry Pi 3, Ubuntu Thinkpad and Windows 10 on original FTDI 232RL not sending

McNugget6750, 2020-02-23:
I am trying hard to get this to work.

I found an original FTDI 232RL in my parts bin, checked the serial number and used FT Prog to try and verify if it's genuine. I cannot find any signs of it being counterfeit. Plus, I bought it way back in 2012 from Digikey.
I have it hooked up to a MAX232 level converter and measure RX and TX using an oscilloscope. I checked everything using HTERM and can open the port and send / receive data there (loopback).

My Apple III is up and running and waiting for data. The inverted "HI" is being displayed in the upper right corner.

I used three different machines with **ADTPro 2.0.3** and earlier versions but I cannot get ADTPro to send any data whatsoever.

**Windows 10 x64:** 
C:\>java -version
java version "1.8.0_192"
Java(TM) SE Runtime Environment (build 1.8.0_192-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.192-b12, mixed mode)

Connection to the FTDI is being established. I can click SOS Bootstrapping and hit ok in the next window. The system then tells me it is sending data but nothing happens on the scope or the Apple III. After about 150 seconds, it says the file was sent. But it clearly wasn't. There was also no character blinking on the Apple III.

**Ubuntu 16.04 64bit on X201 Thinkpad:** Connection to FTDI established. I can click SOS Bootstrapping and hit ok in the next window. The file is being sent in 0 seconds. Nothing happens on the scope or the Apple III.

**Raspberry Pi 3B:** Connection to FTDI established. I can click SOS Bootstrapping and hit ok in the next window. The file is being sent in 0 seconds. Nothing happens on the scope or the Apple III.

I even tried installing various binaries of the RXTX library and made sure they are being used bt ADTPro but no change. I cannot send any data using three vastly different computers.

Btw. Using a counterfeit FTDI shows exactly the same behavior, only this time I am sure it's not a counterfeit chip as stated above.

McNugget6750, 2020-02-23:
The only way I can get ADTPro to send data is to activate the "Imagewriter Cable" pption. But in that case, the III doesn't seem to receive anything.
I also found this thread that seems to add to the story as I'm definitely not the only one with this issue. https://sourceforge.net/p/adtpro/discussion/supportarchive/thread/b0785be9/

[image]

One question that popped up is the wiring: In the image
[image]
You mention the pins that need to be tide together. I wonder, since most of us are using USB2Serial converter modules, these pins are rarely exposed. Is it mandatory to have these pins tide together? Specifically on the PC side? And are these tide together pins then connected to pin 4&20 on the III side? That could explain a lot of things.

McNugget6750, 2020-02-23:
OK, I finally got this to send by taking my clone FTDI and short the pins mentioned above. However, I fear my III is now at fault as it's definitely not receiving anything. I saw the characters blinking through, yesterday, when I sent random characters using HTERM or CUTECOM over the serial port but today, there is no reaction at all.
I'll try and power cycle the III and re-enter the machine code and try again. I hope I didn't fry anything during my experiments.

Please note: The sending is EXTREMELY fast when sending SOS. It takes 0-1 seconds and seems to run at a way too high baud rate. When I go ahead and send something else from your software, it seems to be using a much slower baudrate. Any ideas?

david-schmidt, 2020-02-24:
So far it does look like handshaking is what's going wrong - that is the thing that is going to be helping pace the output.

McNugget6750, 2020-02-24:
I power cycled the III and typed the machine code again. No reaction to any bytes sent anymore (the blinking characters at the top right of the screen when a byte was received). Could I have damaged the III's serial port?

To my above question: only three wires total go between the d-sub 9 and the d-sub 25, correct? The remaining pins mentioned are connected to each other at the respective connector, right?
Also, on the d-sub 9: pins 7&8 are NOT connected to pins 1&4&6, correct?

McNugget6750, 2020-02-28:
Reseating the logic board chips and using an official cable brought communication back to life. ADT started loading but stalled. Turned out both of my Apple /// have memory chips damaged. I'll try to make one working one since both are 12Volts RAM versions.

closed 2020-02-28 by McNugget6750