ADTPro/adtpro #94: 2.0.3. not working with Mac OSX 10.11.6

stratoculumus, 2020-02-23:
I have a fully working ADTPro 2.0.2 (USB-serial and audio) with an Apple II+ and a Mac running OSX 10.11.6 (El Capitan), but for some reason I can't get ADTPro 2.0.3. to work on the same computer. 
Nothing happens when I double-click the app on my Mac. I also tried opening it in the OSX Terminal (command line) by right-clicking on the ADTPro 2.0.3 app, selecting "Show package contents",  dragging the exec file into the terminal window and pressing return, like this:

$ **/Applications/ADTPro-2.0.3/ADTPro-2.0.3.app/Contents/MacOS/ADTPro-2.0.3**

and I get this error message:

Exception in thread "main" java.lang.UnsupportedClassVersionError: org/adtpro/ADTPro : Unsupported major.minor version 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

I'm no programmer, but I'm guessing the problem has something to do with Java? 
I've checked my currently installed Java version like this, in the OSX Terminal:

$ **java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)**

"Software update" on my Mac tells me that I've already installed all available updates from Apple. Is the problem related to something else, or do I have to "manually" install a new version of Java -and if so, where can I download it from?

david-schmidt, 2020-02-24:
So it looks like you're on the right track.  You're in Java purgatory now because you've got more than one level installed - and ADTPro sees the right one (you can point it explicitly at the one you want to use), but Photoshop doesn't know that you have the newer one installed since the "system" JVM is still at that 1.6 level.

stratoculumus, 2020-02-25:
Are you saying that with Java, you can install different versions/types alongside each other, instead of having the latest version replace everything else?
Seeing that "jre-8u241" did no good on my machine I finally found instructions on how to remove it.
I believe I now have two types of Java on my machine: the "legacy" 1.6 (downloaded from one of Apple's sites) which allows Photoshop to run, and Oracle "jdk-13.0.2" which allows ADTPro 2.0.3. to run. Is this the correct way to do it?

I'm probably not the only ADTPro user who finds this confusing and isn't into programming and such: can you consider adding some easy to understand information about all of this in a readme file for ADTPro? And if possible, a clear error message in the app itself (i.e. "Incompatible Java version -please install version xxx or higher") and maybe even a download link for what's needed as all the "SE", "JRE", "JDK" along with the choice between Oracle, Java.com and Apple's own Jave distributions can be quite confusing.

david-schmidt, 2020-02-25:
Unfortunately, it's a fairly deep problem with open-ended possibilities and consequences.  And it's been made harder by Apple stopping support on your operating system some time ago; they've left you to manage Java yourself.  I appreciate that you found it a difficult problem to surmount, and I'm glad you got through it!

closed 2020-02-25 by david-schmidt

stratoculumus, 2020-02-25:
Thanks!
Yes, Apple sometimes like to make things harder than necessary unfortunately. 
I see that there's some mention of Java versions in [the "installation" section](https://adtpro.com/install.html) of the ADTPro website, but could you add that on the Mac platform at least you can (if I've understood it correctly) install both the "legacy" Apple Java version download along with the Oracle one I downloaded, and with links to them?
That could save a lot of users some pain and frustration.