ADTPro/adtpro #96: ADTPro Client on DOS 3.3

0cjs, 2020-03-16:
Is there any way to run the ADTPro client on DOS 3.3?

The scenario here is that I'm doing development on DOS 3.3 and want to frequently back up my work diskette to an image file on the server. (I keep my image files in a Git repo so I can easily go back to earlier versions.) Rather than swapping floppies to the ProDOS ADTPro diskette, rebooting, swapping floppies again, and uploading, I'd like to just have a copy of the ADTPro serial client on my work diskette, `BRUN ADTPRO`, upload, exit, and continue work.

I tried building `ADTPRO.BIN`, putting a DOS 3.3 binary header on it (prefixing a starting address word $0800 and length word $376D, both little-endian), and putting that on my DOS diskette. It appears to `BLOAD` correctly, when I do some quick disassembly checks in the monitor, but when run it immediately drops to the monitor, printing

    FDF7-    A=A5 X=D7 Y=00 P=F4 S=E6

So I guess that program is in some way dependent on running under ProDOS, though I've not dug through the source to see what that dependency is.

david-schmidt, 2020-03-16:
There is not, since ADTPro relies on... ProDOS.

But the ADT client that came before it runs on plain old DOS; you might look into that.
https://github.com/david-schmidt/adt

0cjs, 2020-03-16:
@david-schmidt That did the trick! It seems to work fine with the current ADTPro server in both directions, though I've used only "simple" (non-nybble) transfers. After a download-change-upload sequence with ADT I checked the disk changes by doing a diff of before-and-after dumps made with [`dos33 dump`](https://github.com/deater/dos33fsprogs/tree/master/dos33fs-utils) and they looked correct, with no unexpected changes.

closed 2020-03-16 by 0cjs