ADTPro/adtpro #97: Not able to get ADT Pro to work in Linux, even after viewing http://adtpro.com/install.html

insectorx84, 2020-04-08:
Hi.  Thanks for the awesome software!  I can't wait to use it.  I trying to use a netbook, running Lubuntu 18.04 LTS, to be a dedicated ADT PRO server for my Apple IIgs (windows doesn't run very well on it).
Here is the error I keep getting:
"There are no serial ports, or the serial library RXTX is not installed or visible. Please see http://adtpro.com/install.html for details"
So far, I've gone to some Linux support chat for help, because I thought it was Linux not recognizing my serial adapter. The people on that chat helped me confirm that the adapter was indeed being seen by linux.  We tried a number of other things, all of which didn't work.  

Edit: I have been to http://adtpro.com/install.html, but I'm not getting the same RXTX error that is listed there, even when I have selected "execute from terminal".  Also, I don't understand what it's trying to tell me to change in the document.  Thanks in advance for any help.

PS- cereal adapter.

david-schmidt, 2020-04-08:
There's a couple of things that are going to need to happen - first, we'll need to find out if the binary RXTX library will even run on your particular distro of Linux - and if so, which one it is, and then tune your startup script to use it.  That's mentioned specifically here under "UNIX Considerations": 
https://adtpro.com/install.html#Considerations
(though it doesn't discuss *how* to make that change on a new Linux distro (one that isn't natively supported by ADTPro)).

So the secret is going to be in `adtpro.sh`.  You'll see that it's making decisions based on these three commands:
```
uname
uname -p
uname -m
```
So if you could supply the output from those three commands when you run them from a terminal window, that will shed light on what decisions your script is making.

insectorx84, 2020-04-08:
Thanks the the reply, I really appreciate it!
uname = Linux
uname -p = x86_64
uname -m = x86_64

after opening adtpro.sh I quickly saw where to enter these.  I also saw that there was a spot to enter the location of the ADT PRO folder (mine's on the desktop).   I entered those 3 values in (uname, uname -p, and the path to ADT PRO's folder).  
At first, ADT PRO wouldn't open at all after I made those changes, but I had initially misunderstood how to enter the values. 

I had changed:
OS='uname'
OS-ARCH='uname -p'

To:
OS='Linux'
OS_ARCH='x86_64'

But when I removed the apostrophes the program would open normally again.

Now when I'm opening the program, I'm getting this error in the terminal : "RXTX Warning:  Remove stale lock file. /var/lock/LCK..ttyUSB0".   Also The Bootstrapping Menu is grayed out.  When I click the serial button nothing happens.

Thanks again for the help.
-PVH

david-schmidt, 2020-04-09:
Well, you definitely didn't want to do any of that.  Please revert your `adtpro.sh` shell script back to the way it was originally.  There's a difference between single quotes (') and backtics (`), and they have significance in how they perform in shell scripts.

Once you've done that, what is the output from the script if you run it from the command line?  (i.e. `./adtpro.sh`?

insectorx84, 2020-04-09:
Ok.  My bad.  I replaced the adtpro.sh file with a new one from the tar.gz

I navigated to the ADT PRO folder and opened adtpro.sh using ./adtpro.sh

When I open it, it gives me the same error in the terminal as last time:  
RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyUSB0

(Im down to reinstall a different distribution of Linux if it will work better with ADT PRO.  It just would have to be lightweight for this netbook.  I'm going to make this a dedicated ADT PRO machine, so I don't mind.  I just chose Lubuntu, because I thought it's supposed to be a less resource hungry version of Ubuntu)

david-schmidt, 2020-04-09:
Ok, what if you run it as sudo?  `sudo ./ADTPro.sh`?

insectorx84, 2020-04-09:
running it as sudo gets it to open without that error.  
It displays this in the terminal:
ADTPro Server version 2.0.3

Then it brought up the GUI part of the program

The program is working & actively transfering data to the IIgs!  :) 
But is there an easier way to open this program each time?

david-schmidt, 2020-04-09:
Sure, you can correctly configure your user to have appropriate permissions on your box. :-)

It's possible you actually have residual lock file that need to be removed; you can check out this blog for a similar experience:
https://smdaudhilbe.wordpress.com/2013/05/01/how-to-solve-serial-port-problem-removing-stale-lock-varlocklck-ttyusb0/

Also, adding yourself to the `uucp` group should help:
https://github.com/c-flynn/Nunchuckoo/issues/2

david-schmidt, 2020-05-03:
Closing as any additional details are going to be specific to OP's Linux installation.

closed 2020-05-03 by david-schmidt