ADTPro/adtpro #98: Can ADTPro be used on other 3rd party serial cards for the apple II?

Verault, 2020-05-02:
I work on and restores lots (and I mean lots) of Apple II computers. I come across many different cards, some single function, some multifunction that have serial interfaces. I use ADTPro quite a bit but with Super Serial Cards. Can ADTPro support these 3rd party Serial devices? Has anyone tried. Right now I have this Mountain CPS Multifunction card from 1981 which has a serial port, parallel port, and Real time clock chip with battery. I would love to use ADTPro with this card in an Apple II plus if possible. Would appreciate your feedback on the matter. Here is a link about the mountain card:

https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwjWxY7gvZXpAhUOVs0KHcqQCnoQFjAAegQIAhAB&url=https%3A%2F%2Fmirrors.apple2.org.za%2FApple%2520II%2520Documentation%2520Project%2FInterface%2520Cards%2FMultifunction%2520Cards%2FMountain%2520Computer%2520CPS%2520Multifunction%2520Card%2FManuals%2FMountain%2520Computer%2520CPS%2520Multifunction%2520Card%2520-%2520Reference%2520Manual.pdf&usg=AOvVaw1jYJkoU_v3EEiodRRznvHb

david-schmidt, 2020-05-03:
Hi there - in general, yes, if you have register-level information, that's the interface that ADTPro uses to talk to a serial device.  There's a high-level concept of a "driver" that would need to know how to send and receive a byte including timing control, and then it would need to know how to initialize a card by doing things like setting the baud rate.  The basic concepts are here:
https://github.com/ADTPro/adtpro/blob/master/build/DRIVER.DESIGN

The preferred baud rate is 115.2kbps, which I notice isn't available on the Mountain card.  Coming up with a way to limit or disallow invalid configuration settings would need to be added as well.

michael-jetzer, 2020-09-17:
While I have no information on your Mountain card, to the general topic of third-party serial cards, I can confirm that ADTPro can transfer disk images at 115.2 using the following third-party boards:
- Applied Engineering Serial Pro combination serial and clock card.
- "Super Serial Board SSB-1".  This is a generic SSC clone with a board layout that's the same as a genuine SSC.  Joe's Computer Museum did [a video](https://www.youtube.com/watch?v=V7NtGMPhFNg) on this card.

Verault, 2020-09-17:
Thanks Michael. Yes I also discovered a bunch of different generic SSC clones. They almost all have a complete rip of the Apple SSC down to the ROM code. So essentially they are SSC cards. I just wish there was support for more cards than just the SSC as prices online are going up pretty high (Even for SSC cards without the ribbon cable attachment, I have been having to make my own). Besides the mountain card I hace at least 3 other 3rd party serial cards. I also have The apple Brand serial cards that came out before the SSC and they are NOT supported either which is a shame. I have to figure they sold alot of those.

"""Applied Engineering Serial Pro combination serial and clock card. is supported...""" I think I have one in the box, I need to try that one out.

Thanks
Mick

david-schmidt, 2020-09-17:
> I also have The apple Brand serial cards that came out before the SSC and they are NOT supported either which is a shame. I have to figure they sold alot of those.

ADTcc exists to talk to the Apple Communications Card for those that like to go that slowly.

That said - those that send me hardware (and ask nicely) often get their favorite hardware supported.

Verault, 2020-09-17:
David Completely understandable. Since I have more than one ID be happy to make a donation to the cause if it means others with older cards can utilize the Software (which Is absolutely amazing and breathes all kinds of new life into these "boxes of fun")  Either reach out to me or give me a way to reach out to you so I can send one of these cards in the mail.

Neoncluster, 2021-01-10:
Verault, if you could get David an Apple Communications Card  for testing would be fantastic.
I'd like to see this card working in ADTpro too.

tfrazier-home, 2021-01-10:
I am definitely interested in this topic but not right now. 

I have tried to unsubscribe to this thread several times through the github account to no avail! 

Can someone help me get off this? 

Thanks

Verault, 2021-01-10:
He declined accepting the card. Sorry.