ADTPro/adtpro #99: USB to Serial Woes

RkmanX, 2020-05-06:
At first I could get it to recognize that I indeed have a USB to serial installed on COM 3, then it kept crashing and unstalled the driver and reinstalled it, still crashed. So I uninstalled and tried the non-microsoft provided driver and now it won't read any serial or USB to Serial communication.

I get this, which I guess makes sense if it can't find it. Currently set to COM3. I am beyond confused I am sorry to have to ask for help here. :( This is the only output atm
"
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver" before it when it crashed it output this error log. I am sorry again!
[hs_err_pid23996.log](https://github.com/ADTPro/adtpro/files/4589620/hs_err_pid23996.log)

RkmanX, 2020-05-06:
Alright I have it back to finding the com port, but it is still crashing. I can't run it with command line now as it won't find the com host.

david-schmidt, 2020-05-06:
What USB-RS232 interface are you using?

RkmanX, 2020-05-06:
> What USB-RS232 interface are you using?
I am using a "Trendnet TU-S9" which uses a 'PL2303' chip. Or so the program that comes with the driver says.

RkmanX, 2020-05-06:
Uhm on the newest dump it says 
"# Problematic frame:
# C  [rxtxSerial.dll+0x5b00]"
Is it not able to read my serial connecter?

RkmanX, 2020-05-07:
It just started working uhm. Though I can't seemingly send anything over. I go to send ProDos over to it, it says its transfering the Disk image but it doesnt transfer it. Then once I press disconnect it starts sending garbage data (At least it looks more like garbage than comparaed to the videos I have seen it has tons of spaces makes error beeps etc.) I am so sorry!

scottson2, 2020-08-03:
not working for me as soon as i let java update.  now i can't downrev java back to old version without similar results or worse...

david-schmidt, 2020-08-03:
Yeah, don't upgrade Java right now.  Oracle made a breaking change recently (https://github.com/ADTPro/adtpro/issues/110).  You should be able to find older versions on your machine and modify your ADTPro.bat file to point to the one you have better luck with.

scottson2, 2020-08-03:
I went back to this version and ADTPRO works again...  Uninstall the newly installed version, and then install this version and its back to Apple Land once again.

https://www.oracle.com/java/technologies/javase/javase8u211-later-archive-downloads.html

Windows x64 | 73.29 MB | jre-8u241-windows-x64.exe

david-schmidt, 2021-02-22:
You should also be able to fully upgrade Java along with the latest ADTPro version of 2.1.0, which has better support for Oracle's breaking change.

closed 2021-02-22 by david-schmidt