ADTPro/adtpro #102: ADTPro was working through on IIgs, but won’t work on my IIc

insectorx84, 2020-05-15:
Hi.  I had posted about another issue previously with Lubuntu.  Thanks for your help with that.  The IIgs was working great with ADTPro!

I recently sold my IIgs, and I just got a cable to be able to hook up my new IIc to ADTPro.  Unfortunately, I can’t seem to get it to connect properly.  The side the has my Linux laptop seems to be seeing the connection.  Every time I try to (R)ECEIVE, the Apple IIc side it says: 
Waiting for host reply.
(& then)
Host Timeout 

The Linux side says:
Request:  directory contents

Here is the cable that I’m using,
It’s sold through eBay by a company called
The Apple2Online Store
http://shorturl.at/iptQZ

The UsB to serial adapter I’m using is the same that I was successfully using with the IIgs. 

What I’ve tried so far:  tried both speed ratings under ‘serial configuration’, and have tried them both with the IIc ImageWriter cable option on and off.  

I notice that ‘serial configuration’ Window says that the port is ‘dev/tty/USB1’before I click to connect to serial (in Linux ADTPro).  After I click connect and check the same config screen, the port section is empty.  If I then click ‘ok’, it then gives me the error:
“There are no serial ports, or serial library RXTX is not installed or visible. Please see http://ADTPro.com/install.html for details“

david-schmidt, 2020-05-16:
Hmmm, all those symptoms don't line up.  If you are getting the "There are no serial ports, or serial library RXTX is not installed or visible" message, then that's exactly what it means: the USB adapter is not getting seen by the server app.  So this exact setup couldn't have been working with the IIgs, and the server side wouldn't have changed when you swapped that for the IIc and the IIc cable.  So is the USB-RS232 adapter the same one you used with the IIgs, or not?

insectorx84, 2020-05-16:
Im definitely using the same usb to serial adapter as i was before on the
IIgs (I only have one).  And it seems that ADTPro sees the usb adapter when
i open the program, but then it disappears (under “ports” in config) after
i click “connect”.  ( I get the error message somewhere in the config menu
on the Linux machine, btw)

After I got it working with the IIgs, I hadnt messed with it until
yesterday with the IIc.  The only setting that i changed was the bps speed
and the check box for the ImageWriter (while trying to get it to work).
Any time i changed the speed, I mirrored the changed in adtpro on the apple
ii.

Unfortunately I dont have the IIgs anymore to confirm that its still
working there.  Had to sell to make rent last month. Lol

Thanks for your time. :)
-Pete

david-schmidt, 2020-05-16:
The disappearing act is definitely a problem.  Maybe try without your IIc plugged in, and see if clicking "connect" still makes it go away.  If so, let's have a look at _all_ the text that is in the console window at that time - which I expect will be some Java dump information.

insectorx84, 2020-05-16:
Here is the Java dump when I tried to connect without being plugged into the IIc.  I still have the USB to serial adapter plugged in.

gnu.io.NoSuchPortException
	at gnu.io.CommPortIdentifier.getPortIdentifier(CommPortIdentifier.java:269)
	at org.adtpro.transport.SerialTransport.open(Unknown Source)
	at org.adtpro.transport.SerialTransport.open(Unknown Source)
	at org.adtpro.CommsThread.<init>(Unknown Source)
	at org.adtpro.gui.Gui.startComms(Unknown Source)
	at org.adtpro.gui.Gui.actionPerformed(Unknown Source)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:401)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6631)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6396)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5007)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4839)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4839)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

david-schmidt, 2020-05-16:
It's definitely crashing that thread when it tries to talk to the port.  Try restarting the app, and watch the output window when you go to select a port the first time - note what's in the list (I suspect USBxx like before) and then select it and go back in a second time.  Is it still there?  Then, try the "connect" button (which I expect will crash like above).

insectorx84, 2020-05-16:
(I'm still doing this without the IIc connected, but I still have the usb adapter plugged in)

/dev/ttyUSB0  is the port's name
When I open the app and go to select that port, I get this error in the console:
RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyUSB0

I went back into the serial config menu (without closing and reopening the ADTPro) and the port still shows up as /dev/ttyUSB0 under ports

david-schmidt, 2020-05-16:
Ah, there's your clue... you have a stale lock file.  ADTPro has been running as root and you're no longer root.

insectorx84, 2020-05-16:
I'm not sure what that means tbh.  I'm starting up ADTPro from the command line with sudo ./adtpro.sh 
I remember that was what you had me do to get it running for the IIgs, was to use the sudo command.

david-schmidt, 2020-05-16:
Does it complain about `/var/lock/LCK..ttyUSB0` every time?  Is that file still there after you close the ADTPro server?  If so, can you `sudo rm` it?

insectorx84, 2020-05-16:
I tried the same thing again and it didn't complain about it this time.  I closed adtpro then went and sudo removed the /var/lock/LCK..ttyUSB0 and restarted ADTPro, this time connected to the IIc and got the same results.  Host timeout on the IIc end, while ADTPro on the Linux side shows that im requesting directory contents
While it was open i got this error in the console when I tried to check my port (which was missing btw).:   RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyUSB0: File exists

david-schmidt, 2020-05-16:
Hmmm.  Definitely something going on with your permissions and the USB port.

insectorx84, 2020-05-19:
Do you think this is a problem with my OS or my computer?   Im down to
replace either or both if thats what it takes.

david-schmidt, 2020-05-19:
If you're not a *nix guru, it probably would be helpful to have a more common OS.

insectorx84, 2020-05-19:
Ok.  Im definitely not a *nix guru, but im trying to figure it out slowly
project by project.  If theres a smarter way to learn it then Im totally
down for any advice. :)

  I guess i should have used the word Distro instead if OS since I dont
have an available modern laptop with windows or MacOS X.    I think what
I’ll do is keep trying different combinations of older laptops with
different distros of Linux until I come up with a working combo.  Hopefully
that'll help get it working again.

 (Id love to use this netbook as a dedicated adt machine since its so tiny
and convenient, but convenience isn't worth much if it wont work for me.
Lol)

Ps- (Im also trying to mess with MacADT although I've got to find a Dos 3.3
boot disk somehow i think)

insectorx84, 2020-05-24:
Ok, I'm trying to get this set up with a different distro.  Your website says that I need Java, but now I see there is multiple versions available.  Is the open source version that comes preinstalled with ubuntu ok, or do i need to go find a way to install the official one from oracle?

PS - I don't have a windows machine available to do this so it'll have to be linux.

insectorx84, 2020-05-24:
ok, so I've got a new distro installed and when I start up adtpro I'm getting the exact same results.  Maybe it's possible that I'm setting up adtpro wrong?

Here is what i've done after installing the new distro (I've done literally nothing else to this system)
-update/upgrade
-copy over my adtpro folder from usb
-install java
-install rxtx
-add myself to 'dialout' group
-check groups to confirm I'm in dialout
-run 'sudo ./adtpro.sh' from the terminal-
-then i get same problems as above

The distro i'm using is Based on Ubuntu 18 called Linux Lite.  From what I understand, it's Ubuntu with a custom desktop environment (so it's super lightweight)

insectorx84, 2020-05-26:
Ive just tried this on another computer and had the same results.   It
seems like im Possibly having one of the following problems:

-Im maybe not installing everything right.
i.e.- Java or rxtx maybe.?
- my Din5 to rs232 cable may be bad?
-my iic may be the problem Somehow?

I come to that conclusion, because I've tried multiple linux distros on
multiple computers with multiple usb to serial converters  So the only
things left are the possible options above ( i feel)

Ps- Im also not able to get MacADT to transer dos or adt to the apple iic
no matter what setting I use.   Maybe this points to there being another
problem altogether??  I dont know.

insectorx84, 2020-06-01:
Ok so I finally got the chance to hook this up to a windows computer and I'm getting the same results unfortunately.

insectorx84, 2020-06-03:
Well, I guess it was the cable after all.  I was able to get it to work with (of all things) an Imagewriter II cable, and a Keyspan usb to miniDIN8 converter (Mac Only, Model USA-28x) I had laying around.  I would have never guessed that Linux could push that converter without a proper driver.  But now I've got to order a new cable from retrofloppy.com to make sure it works properly.  Thanks for all the help, even though the problem ended up being something simple, I learned a lot about Linux and ADTPro during the adventure to figure it out.  Thanks again.

closed 2020-06-03 by insectorx84