ADTPro/adtpro #103: ADTPro and Tiger Learning Computer?

DeathAdderSF, 2020-05-29:
Howdy, all.

I regularly use ADTPro w/my IIc and IIe; what a lifesaver! So, many thanks for making the program available to us. :)

I also have a Tiger Learning Computer, which is based on the IIe and running ProDOS, which I'd like to get connected with ADTPro so I can archive its built-in software, and to get programs to and from the computer for general research purposes. Buuuuut, I haven't had any luck getting it to bootstrap after well over a dozen different attempts.

I'm hoping someone may be able to get in touch with me directly, so we don't bog down the issue tracker here, to see if we can work together to find a solution. If we have a breakthrough I'm happy to share any software, and info I'm able to glean from the experience.

Please feel welcome to get ahold of me by e-mail...

getpsyched@diskman.com

Thanks in advance. Have a nice day.

david-schmidt, 2020-05-29:
I also have a TLC - I have not had any success getting data in or out.  I don't think there's an awful lot to talk about unless you have?  Bootstrapping is a distant next step after we can understand how the serial port works (if that's even what it is).

DeathAdderSF, 2020-05-29:
Thanks for the speedy reply, David. If you've had no success either, you're right: there's nothing more to say for now. Here's hoping more info on the hardware surfaces in the future.

closed 2020-05-29 by DeathAdderSF