ADTPro/adtpro #105: Speediboot speed issues

thenebular, 2020-06-04:
I've been trying to get ADTPro to work properly on my iigs for quite some time. I've had to build my own serial adapter, but since I don't have a DIN-8 plug I can use, I've been going through a breadboard, so for most of the time I've been trying to diagnose issues with my cable. Basically, nothing would send over to the Apple. Then the wire for pin at on the apple serial port accidentally disconnected and everything started to work until it got to "Found Slot 8" "Loading MLI" then nothing. I went through the old method of loading ProDos then ADTPro, but I would still be getting timeouts. Then I tried just getting rid of handshaking all together and basically made a iic type adapter with pins 1 & 2 on the apple tied together. Same exact issues as before, however this time I tried changing my serial speed in ADT to 19200 and everything worked. So either my Toughbook I'm using for the server doesn't like it faster that 19200 or the Apple, I don't know, but after looking through the trace I noticed that whenever using speediboot, it would change the port speed to 115200 once it started looking for PDspeed.raw no matter what my settings are set to.

I don't have a working floppy drive so I need the virtual drive, but I don't know how to load it without speediboot and I don't know how to make speediboot use the port at 19200.

So any help would be appreciated.

david-schmidt, 2020-06-04:
Speediboot will give you trouble since it is going to demand to use the built-in hardware at the built-in supported speed.  Since your cabling situation isn't letting you use the full speed your machine is otherwise capable of, you're going to have to do everything manually - see this section in the doc:
https://adtpro.com/bootstrap.html#If_Speediboot_doesnt_work_for_you
The old-style bootstrapping video is re-enabled in that section now (it had disappeared at some point) and it should help as well.

thenebular, 2020-06-04:
Ok, So to get speediboot working I'd need to somehow get windows to report that the serial port can only support 19200. But since I don't have a floppy drive to work with how do I get the virtual drive working doing everything manually?

david-schmidt, 2020-06-04:
You have too many holes in your hardware, unfortunately.  You need a good serial connection on the modem port, or you need a floppy drive to load up a VSdrive disk.  Or both.  But you have neither, so I'm afraid there's not a path forward.

thenebular, 2020-06-04:
Could the Ramdisk be a solution?

david-schmidt, 2020-06-04:
No, that doesn't help because booting over serial avoids all disk access.  If you can boot from a physical floppy, you don't have to use serial and can make configuration changes later.  But the RAMdisk doesn't help in either scenario.

thenebular, 2020-06-08:
Ok, well I swapped out my toughbook for a desktop and everything went fine with my existing homebrew cables, looks like the CF-19's port was the problem.

So speediboot+virtual drive boots to a command line. Do I need to send the ADTPro serial to get into it or is it already available and I just need to run a command? (This is my first Apple since my elementary school's ones and I don't know a whole lot about the command line yet)

david-schmidt, 2020-06-08:
You're back in a catch-22 situation, unfortunately.  What VSdrive gives you is the ability to do virtual disk I/O over the serial port; but that function is separate from and mutually exclusive with the ADTPro whole-disk image transfer function.  If you want to use a virtual serial drive to do your own I/O with, that's what VSdrive is for.  If you want to transfer full-disk images, that's what ADTPro is for.  But it seems to me that you want to somehow have both - i.e. use ADTPro to transfer virtual floppy disk images to a virtual serial disk.  And that is just circular.

There are some limited situations where you could set up VSdrive this way and serve some ProDOS images to your Apple II - but almost none of those will be the full-function floppy disks such as games.  So maybe it's time we step back and find out what your goals with this extremely limited hardware are?