ADTPro/adtpro #106: Linux adtpro.sh gives errors

16csvt, 2020-06-14:
I have downloaded and uncompressed the linux tarball.
I get errors when adtpro.sh executes.
I have looked in issues, both open and closed and cannot find any issue that may suggest a cause / cure with the symptoms I have.
One thread did ask a number of questions, and I have pre-empted those questions by copy and paste from the terminal window.
Can anyone please suggest what I might try next to resolve this matter.
Regards
Kevin

[khand@localhost ADTPro-2.0.3]$ uname
Linux
[khand@localhost ADTPro-2.0.3]$ uname -p
x86_64
[khand@localhost ADTPro-2.0.3]$ uname -m
x86_64
[khand@localhost ADTPro-2.0.3]$ ./adtpro.sh 
ADTPro Server version 2.0.3

Exception in thread "main" java.awt.HeadlessException
	at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)
	at java.awt.Window.<init>(Window.java:536)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at org.adtpro.gui.Gui.<init>(Unknown Source)
	at org.adtpro.ADTPro.main(Unknown Source)
[khand@localhost ADTPro-2.0.3]$ ./adtpro.sh serial
ADTPro Server version 2.0.3

Exception in thread "main" java.awt.HeadlessException
	at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)
	at java.awt.Window.<init>(Window.java:536)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at org.adtpro.gui.Gui.<init>(Unknown Source)
	at org.adtpro.ADTPro.main(Unknown Source)
[khand@localhost ADTPro-2.0.3]$ su
Password: 
[root@localhost ADTPro-2.0.3]# ./adtpro.sh serial
ADTPro Server version 2.0.3

Exception in thread "main" java.awt.HeadlessException
	at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)
	at java.awt.Window.<init>(Window.java:536)
	at java.awt.Frame.<init>(Frame.java:420)
	at java.awt.Frame.<init>(Frame.java:385)
	at javax.swing.JFrame.<init>(JFrame.java:189)
	at org.adtpro.gui.Gui.<init>(Unknown Source)
	at org.adtpro.ADTPro.main(Unknown Source)
[root@localhost ADTPro-2.0.3]# exit
exit
[khand@localhost ADTPro-2.0.3]$ java -version
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)
[khand@localhost ADTPro-2.0.3]$

david-schmidt, 2020-06-15:
It's complaining that it can't find a graphics context.  What Linux distro are you using?  Is it in fact headless?

16csvt, 2020-06-15:
Distro is Fedora 31 (due to up to 32, but that now on hold pending resolution)
Celeron cpu G3930 @ 2.9 GHz x2
Intel HD Graphics 610 (Kaby Lake GT1)
Gnome 3,34,4
64 bit
The system is NOT running headless.

Is there a relatively quick and dirty way of checking the environment java thinks it is in?

schmidtd, 2020-06-15:
Your jdk/jvm is headless.  Using your package manager, query for `headless` and you may find that's true.  One example I saw here shows some causes and solutions:
https://lists.fedoraproject.org/archives/list/users@lists.fedoraproject.org/thread/S7QBL3UHYBAXWUMQJTMASGIYTT237AWH/

david-schmidt, 2020-06-15:
Your jdk/jvm is headless. Using your package manager, query for headless and you may find that's true. One example I saw here shows some causes and solutions:
https://lists.fedoraproject.org/archives/list/users@lists.fedoraproject.org/thread/S7QBL3UHYBAXWUMQJTMASGIYTT237AWH/

16csvt, 2020-06-15:
Thank you for the advice.
Once I had a clue as to the issue, google soon revealed that it seems that the java as installed  in this distro is indeed headless, and has been for a number of Fedora versions.
dnf install java pulls 30 odd meg of additional packages, and I now see the panel for serial / ethernet/ audio. There remain issues relation to users, groups, permissions and lockfiles, which I will now address. These issues are not however related to the main issue in this thread, which I will now try to mark 'Solved' and close it.
Many thanks.