ADTPro/adtpro #109: Windows 10 ADTPro server binding to 172.x.x.x address vs 192.168.0.x

glonch, 2020-07-22:
Greetings all,

When I click on ethernet in the main ADTPro window, it is binding to a 172.x.x.x address.

My network is 192.168.0.x

Any ideas?

Thanks

Rich Thompson

johnddias, 2023-03-13:
Leaving this for others, hopefully not too late for Rich :) 

I'm running ADTPro on Windows 10 and I had HyperV enabled, which created a default switch. That's where my 172.x.x.x network address was coming from. 

To solve this, I went into the Advanced TCP/IP properties of the vEthernet Hyper-V adapter and unchecked "Automatic metric" and set the value to 100. I then went into my 'regular' Ethernet adapter and unchecked "Automatic metric" and set the value to 50 (lower takes precedence). That fixed it for me.

glonch, 2023-03-13:
Thanks for responding!  I did figure it out but I completely forgot how I did.  lol