ADTPro/adtpro #110: Java 1.8.0_261-b12 on Windows breaking JNI change, crashes on connection to RXTX

david-schmidt, 2020-07-27:
A new breaking change on Windows JRE 1.8.0_261-b12 causes a crash when connection to RXTX through the Java JNI.  The symptom is the main GUI window disappears when clicking on the "Serial" button.  Remediation in the mean time is to modify the `adtpro.bat` file to point to any prior version of Java; for example:
`SET MY_JAVA_HOME=c:\Progra~1\Java\jdk1.8.0_91\bin\`

johnsonlam, 2020-07-27:
Hi David,

Even I add the "SET MY_JAVA_HOME=C:\java\bin\" to ADTPRO.BAT didn't work.

A file "hs_err_pid1176.log" created in the ADTPRO folder:

[hs_err_pid1176.log](https://github.com/ADTPro/adtpro/files/4980957/hs_err_pid1176.log)

I've give up ADTPRO for the time being, until you have some way to fix. I also tried with TeraTerm PRO and know the connection really established, I can send keystroke to Apple, but what command did I need to send to the Apple? And how to run the disk image? Just CALL-151? Please kindly advise, thanks.

david-schmidt, 2020-07-31:
@johnsonlam - You need to set `MY_JAVA_HOME` to a place that actually has a different version of Java than 1.8.0_261-b12.

Samuel-Oliveira, 2020-11-12:
JRE 1.8.0_271 get this error too. Sound like Java change something and will not come back. Someone know how to rebuild the DLL?

leebehrens, 2020-11-12:
This appears to be the latest source, with most recent update 9 Apr 2015: https://github.com/rxtx/rxtx
Based on timestamps (07 Dec 2008), the compiled version included in ADTPro seems to have come from http://fizzed.com/oss/rxtx-for-java

leebehrens, 2020-11-12:
> JRE 1.8.0_271 get this error too. Sound like Java change something and will not come back. Someone know how to rebuild the DLL?

This appears to be the latest source, with most recent update 9 Apr 2015: https://github.com/rxtx/rxtx
Based on timestamps (07 Dec 2008), the compiled version included in ADTPro seems to have come from http://fizzed.com/oss/rxtx-for-java

david-schmidt, 2020-11-17:
There's good news on the horizon - I've been able to drop in `jssc` (https://github.com/java-native/jssc) and it's working in initial tests.  I have some performance tuning and a lot of testing to do, but it was overall a very painless transition to enumerating and using physical serial ports.  This will free us from `rxtx` which seems to not be maintained, and the distribution footprint is far simpler.

david-schmidt, 2021-02-22:
This is implemented and working in the field now with some good history behind it, so closing this issue.

closed 2021-02-22 by david-schmidt