ADTPro/adtpro #111: Disconnects / overrides MicroDrive Turbo

ewindisch, 2020-07-28:
Every time I connect with VEDRIVE, I lose access to the partitions of my MicroDrive/T. Attempting to read from those drives freezes PRODOS 8.

This was not a problem before installing the MD/T.

My configuration:
 IIGS ROM1
 RAM 8MB
 Slot 1 - Microdrive/Turbo (also tested slot 2)
 Slot 3 - Uthernet 2

ewindisch, 2020-07-28:
Reading through #29...

ewindisch, 2020-07-28:
From what I can tell, I seem to be experiencing #29 despite running 2.0.3?

Reordering the slots works, but that has cost me both serial ports as I need Slot 7 for Appletalk.

david-schmidt, 2020-07-28:
Hmmm, you’re really testing the outer limits of this use case there.  Have you saved a configuration for VEDRIVE with the slot for the Uthernet defined? (I’ll have to check if there’s a similar bug in it where it scans anyway)

ewindisch, 2020-07-28:
My vedrive.config contains "110.10.0.130" (where the "10.10.0.130" is an IP address)

The configuration did not prompt or ask me for a slot number. I do not have adtpro itself in the same directory, so I have no configuration for adtpro. Should I also have a copy if my adtpro configuration in the vedrive directory?