ADTPro/adtpro #116: Startup problems on Windows 10

joegreaves, 2020-09-13:
I am new to this but an old Apple II user since 1979 and just pulled some old machines out of storage and wanted to try this. I have a Core i5 based system with a serial port and a null modem cable and a fresh Win10 Pro install and loaded the Java program and then try to run the ADTPRo 2.0.3 and I get the following error log when I try to run it:  

ADTPro Server version 2.0.3

Exception in thread "main" java.lang.NumberFormatException: For input string: "56#"
        at java.lang.NumberFormatException.forInputString(Unknown Source)
        at java.lang.Integer.parseInt(Unknown Source)
        at java.lang.Integer.parseInt(Unknown Source)
        at org.adtpro.gui.Gui.<init>(Unknown Source)
        at org.adtpro.ADTPro.main(Unknown Source)

I am able to get the graphical window for version 1.1.6 and it detects the COM1 serial port for 115K baud but when I click on serial port, it crashes.

Is there some ADTPRO for dummies startup instructions?  The motherboard is an Asus model and this should not be this hard.

Joe Greaves

david-schmidt, 2020-09-13:
Let’s start with your java level.  You said you have/had version V1.8.261, which is known to be bad.  You should have then uninstalled that and installed something earlier - V1.8.*something.  So I’d be interested to know what version is on your box now.  

Next, for heavens sake, why did you go so far back in time to use 1.1.6 of ADTPro?  Definitely don’t do that, and ensure you don’t leave an old adtpro.properties file lying around, which is probably what is causing 2.0.3 to choke on a non-integer...

david-schmidt, 2020-12-15:
This should be able to be resolved by using version v2.1.0 - please try that and re-open if you still have trouble.

closed 2020-12-15 by david-schmidt