ADTPro/adtpro #117: ADTPro 2.0.3 thinks all my COM ports are in use!

Calab, 2020-09-23:
I'm running Windows 10 64bit with ADTPro v2.0.3. I have 4 COM ports. 1 is the mainboard COM port, 2 and 3 are on a PCIe expansion card, the fourth is COM5 from a USB dongle used to talk to old Atari computers.

No matter which COM port I select in ADTPro it tells me that the port is in use. I have verified that no other software is using the port. The ports all look fine when looking in the device manager.

I have my ADTProTrace.txt file, but GitHub won't allow me to attach it.

Can someone suggest why ADTPro thinks all my COM ports are in use?

Calab, 2020-09-23:
Attached is my ADTProTrace.txt file.

[ADTProTrace.txt](https://github.com/ADTPro/adtpro/files/5265514/ADTProTrace.txt)

david-schmidt, 2020-09-23:
Your system is in fact reporting that the port is in use in the trace you attached.  To make absolutely sure it's not, you might try unplugging and re-plugging the USB dongle and see if that makes a difference.

Calab, 2020-09-23:
I unplugged the USB dongle and COM5 disappeared as expected. No change otherwise.

I disabled COM2 and rebooted, then enabled COM2. Now ADTPro is crashing when I try to start the serial connection. Log is attached.

[hs_err_pid552.log](https://github.com/ADTPro/adtpro/files/5268754/hs_err_pid552.log)

Calab, 2020-09-23:
Followup... I rebooted and the first time I tried to connect using COM2 it said the port was in use. I tried to connect to COM2 again and ADTPro crashed again.

david-schmidt, 2020-09-23:
Ah, you're hitting the JNI failure in issue #110 .

closed 2020-09-23 by david-schmidt