ADTPro/adtpro #119: ADT Pro does not work

pc-bastler, 2020-11-23:
Hi,

I am trying to use ADT Pro, but cannot get it to work at all (tried version 2.0.3, 2.0.2, 1.3.0 with the same result):

I start ADTPRO.command on my Mac (Tried JDK 15 and JDK 8u271), the server starts up. I copied the RXTX-files as described in the readme, set the server up to use my USB CH340 (/dev/tty.usbserial-FD430), tried with and without the option Apple IIc ImageWriter Cable, tried 115200 and 19200, also tried /dev/cu.usbserial-FD430 all with the same result:

Trying to bootstrap "Speedyboot" ends at "Loading MFI:", bootstrapping ProDOS then the serial client afterwards works, but each try to download an image from the Mac ends up in "Host timeout" shown on the client.

I also tried to get it working with a windows laptop as the server, but with that machine, I don't even get the bootstrap to work. Tried with several Java Versions.

Mac is a Catalina 10.15.6 Maschine, Client is an Apple IIc, the cable used is custom built as shown on the website, USB-Serial adapter used is an CH-340.

leebehrens, 2020-11-23:
On Windows you need to run it with a version of Java no older than 1.8.0_202. This should get you going again: https://github.com/ADTPro/adtpro/issues/110#issuecomment-726127376

david-schmidt, 2020-11-24:
On your Mac, first of all - you'll want to undo absolutely everything you did regarding `rxtx`.  Do not copy anything anywhere.  All you need is all in the distribution as-is.

Next, don't change any of the defaults for speed.  The defaults work at the highest speed unless something else is wrong - and changing the speed won't help that.  So don't change them.

Your CH340 driver may be bad, or this may not be an appropriate device for high speed serial comms.  I don't use them, nor do I recommend them.  FDTI is basically the only way to go these days.

All that said: try turning on tracing: File->Activate Trace and retry your bootstrapping operation.  I've found some timing issues specific to MacOS as I'm moving away from `rxtx` and adopting `jssc` for serial communications.

pc-bastler, 2020-11-24:
> On Windows you need to run it with a version of Java no newer than 1.8.0_202. This should get you going again: [#110 (comment)](https://github.com/ADTPro/adtpro/issues/110#issuecomment-726127376)

I also tried 1.8.0_152 with no luck.

> On your Mac, first of all - you'll want to undo absolutely everything you did regarding `rxtx`. Do not copy anything anywhere. All you need is all in the distribution as-is.

OK, I will remove the two files I copied to the extension dir.

> Next, don't change any of the defaults for speed. The defaults work at the highest speed unless something else is wrong - and changing the speed won't help that. So don't change them.

I thought so, I changed the speed after I tried with full speed.

> Your CH340 driver may be bad, or this may not be an appropriate device for high speed serial comms. I don't use them, nor do I recommend them. FDTI is basically the only way to go these days.

I was afraid that this could be the problem... I will look for a FDTI one and try with it.

> All that said: try turning on tracing: File->Activate Trace and retry your bootstrapping operation. I've found some timing issues specific to MacOS as I'm moving away from `rxtx` and adopting `jssc` for serial communications.

I will let you know, if it works as soon as I get an appropriate serial adapter.

pc-bastler, 2020-11-24:
Thank you for you support... after deleting the files and changing to a FDTI Adapter 2.0.3 works great.

closed 2020-11-24 by pc-bastler