ADTPro/adtpro #120: ADTPro will communicate but not function.

blainfaulkner, 2020-11-26:
Back Story,

I am relatively new to the ADTPro scene.  I have a background in IT but not in development.

The situation.

I have tried on several Mac computers to get ADTPro to work with no luck.  I have a provided ADTPro 2.0.3 disk purchased at the time of buying the supplies for this project.  The computer in question is an Apple //c with ROM version 3.  
The cables in use were purchased from apple2online.com.  I am using the null modem cable they offer with an FTDI-FT232RL usb to rs232 adapter. 

The long and the short is this; on every version of macOS (10.6-11.0.1) that I have tried I get the following symptoms.

Launch ADTPro 2.0.3 without issue using the ADTPro.command shell script.
Connect with the serial option with all default setting selected. (I have messed with them in an attempt with no change, thus I reverted back)
ADTPro Server side shows my FTDI adapter as the connected port, No activity above the progress bar, and connected below the progress bar
Then on the //c side of the world with serial selected at first boot I attempt to view the directory.
The //c reports Waiting for the host to reply, followed by host timeout.
On the server side of the world when I do this I see in the GUI "Request: directory contents" and below the status bar "Image Transfer aborted" is shown momentarily followed by just the path to the directory.
In my Java terminal windows I see no activity.

The fact that this is consistent across multiple OS's indicates user error but I just can't figure out what I am doing wrong even after reviewing the documentation.
I have provided images below incase my descriptions above are not clear.

[image]
[image]
[image]
[image]
[image]

Thanks so much in advance for any help you can provide.

david-schmidt, 2020-11-26:
This is almost certainly a problem with your IIc serial cable.  One thing to try - on the server, click on File->Serial Configuration, and on the dialog box that comes up, tick the "Apple IIc w/Imagewriter Cable" and see if that changes anything.

blainfaulkner, 2020-11-26:
Thanks you so much for looking at my post I greatly appreciate it.

I attempted the  "Apple IIc w/Imagewriter Cable" before and just failed to mention it. 

I will add however in the Java terminal window I was able to get some information to show up that might be relevant.  Take a look at the screen grab and let me know what you think if you don't mind.

This happened with all default settings and on both my macOS 10.6 machine and my macOS 11 machine.  I had gotten this before but it had stopped for some reason and figured it was due to me messing around with the settings but it is consistent at this point with the defaults. I have also attached the trace log as well if that might give more detail.

I am open to the possibility that it is the cable I just want to make sure that is the issue before I order a new one.

Thanks again.
[image]
[ADTProTrace.txt](https://github.com/ADTPro/adtpro/files/5604675/ADTProTrace.txt)

david-schmidt, 2020-11-26:
That's consistent with wonky handshaking lines - we just aren't getting good 2-way communications.  It's also possible that there's a hardware fault in the IIc, but I find dodgy wiring to be far more common.  When you hear hoof beats, think horses, not zebras and all that.

blainfaulkner, 2020-12-02:
Hey guys,

Sorry for the radio silence, I had to order parts.

So David hit the nail right on the head.  Turns out the cable I had ordered from apple2online.com was the issue.  I don't want to sling mud but my experiance with those guys was not good right down to what they ended up providing me.

To solve my issue I just ordered a fresh cable from Retro Floppy, smoothing sailing from that point forward.  I literally just pulled it out of the package and I was office to races.

Thanks again for the quick responses and hopefully this will help someone else that has this issue down the line.

david-schmidt, 2020-12-02:
You got the right tool for the job.  Glad you got up and running - enjoy!

closed 2020-12-02 by david-schmidt