ADTPro/adtpro #122: ADTPro Client locks up

mikepolzin, 2020-12-14:
I am having issues getting ADTPro Serial Client to run. When I attempt to do Speediboot, I can see a bunch of data moving, but at the end it just locks up and sits there.

In an attempt to diagnose the issue, I moved to the old method and sent over ProDos, which seems to work fine as it displays the ProDos screen on completion of the load and leaves me at a prompt. Then I moved on the sending over the ADTPro Serial Client, and again I can see all of the data moving, but when it gets to the end it locks up.

I am at a bit of a loss, as obviously the transfer is working. 

I am using a Super Serial Card attached to a straight through cable, with a USB to RS232 adapter that used the FTDI chipset. I am running on Windows 10.

david-schmidt, 2020-12-14:
It sounds like things are going bad when the client starts up, which could indicate a problem with devices that are connected but aren’t talking.  Does it behave the same way when you unplug everything but the Super Serial card from the Apple II?

mikepolzin, 2020-12-14:
So are you saying to pull all cards from the computer except the Super Serial card, including the disc controller? Otherwise the only other external devices I have hooked up are the drives, and I do have two sets of those (one as a back up set).

david-schmidt, 2020-12-14:
Yep - eliminate everything that is not the apple II and the comms link. If things come up smoothly, then you know what to suspect next.

mikepolzin, 2020-12-14:
So other and the Super Serial card I have an 80 column RGB card and the apple disc card. Initially I pulled them both, and the ADTPro Client loaded. I replaced the 80 column card and tried again, and the ADTPro client loaded. Then I inserted the disc controller and I was back to the same situation.

Then I pulled the 80 column card but left the the disc controller card in and the ADPPro client loaded, so the disc controller and 80 column cards must not be playing nicely together. I have no idea why, but at least I was able to get a disc formatted. Tomorrow I will try and get some software written to that disc.

mikepolzin, 2020-12-14:
I was able to go through the process of receiving ADTPRO-2.0.3.DSK and it said that it was successful. I did a (R)eceive, then typed in ADTPRO-2.0.3.DSK, however when I reboot the computer I am getting the following message centered on the screen *** UNABLE TO LOAD PRODOS ***

I tried swapping to the other et of spare drives that I have and I am getting the same message. Any ideas on what might be causing this?

david-schmidt, 2020-12-14:
All of your symptoms are pointing to a hardware problem in your Apple II, most likely with memory.  You didn't mention the kind of machine you have - there are some pieces of advice specific to the non-e versions, but if it's a IIe, your options are slimmer.  If you can recreate your boot scenario and download a diagnostic disk of some kind, that might help - and there's a key sequence to hold down on power-up to initiate a self-test if you've got a IIe.

mikepolzin, 2020-12-14:
I am using an Apple IIe. I pulled Tapper from a site and wrote that to disk, and while it took a long time to load, it did work and play. I also tried the Open Apple + Closed Apple + Reset built-in test, and that came back with "KERNEL OK" I will have to go looking for one of these diagnostic tools to see if I can dig deeper. I did a quick search and seen Apple IIe Diagnostic and Apex II recommended. I am not sure if one is better than the other.

david-schmidt, 2020-12-14:
ProDOS is a pretty good acid test for memory - and your machine is failing.  That's essentially what that "UNABLE TO LOAD PRODOS" is all about.

mikepolzin, 2021-01-13:
It was not a memory issue. I was able to get Apex 2 onto a disk and run that and everything on the Apple IIe checked out perfectly. It turns out it was old floppy disks. When I scanned the floppies in Apex it identified several bad sectors on the disk. I re-formatted the disk several times to recondition it, and brought down ADTPro again and it boots perfectly fine. I have a 1000 35+ year old 5.25 disks and it appears they aren't holding up well.

closed 2021-01-13 by mikepolzin