ADTPro/adtpro #123: bootstrap process hangs on apple//e after grub2.dmp is sent

highflux7, 2020-12-24:
Running ADTPro server on Linux (Ubuntu 18.04)
openjdk 11.0.9.1 2020-11-04
OpenJDK Runtime Environment (build 11.0.9.1+1-Ubuntu-0ubuntu1.18.04)
OpenJDK 64-Bit Server VM (build 11.0.9.1+1-Ubuntu-0ubuntu1.18.04, mixed mode, sharing)

Serial connection to Apple //e super serial card.  Follow instructions to the letter.  Attempt to bootstrap Speediboot followed by Send ProDOS and Send ADTPro Serial Client.

Speediboot transfers quickly, and the Apple//e displays:

ADTPRO SERIAL BOOTSTRAPPER
FOUND SLOT 2
LOADING MLI:

At this point, no more transfer occurs. No blinking cursor/star after the MLI: prompt, nothing.  Basically the apple is locked up.
On the Linux server end, the client seems satisfied that Speediboot was sent and is ready to send more.  If I choose 'Send ProDOS', it instructs me to again do the IN#2, <ctrl-a>14B on the apple to prepare to receive, though which can be done, will eventually fail after receiving screenfulls of data.  If done at the MLI: prompt (from above), nothing happens; the server sends data, but the apple doesn't seem to be receiving anything.  Same result(s) if attempt to 'Send ADTPro Serial Client' is chosen, either via the instructions (IN#2, <ctrl-A>14B) etc, or the MLI: prompt.

What is going wrong??  I had no problems several years ago using the MS/DOS server version of ADTPro.  Nothing of significance is different here other than linux instead of MS/DOS and ADTPro/Java.  Obviously data is being transferred as at least part of the bootstrap gets over although trying to bootstrap ProDos or the ADTPro serial client directly crashes.  Serial speed is set at 19200.

david-schmidt, 2020-12-24:
So, absolutely everything of significance has changed since you used the MS-DOS version of ADT (which is not the same as ADTPro).  Absolutely everything except maybe your cables.

You didn't follow the directions to the letter, because nowhere in the directions does it say to change the defaults for transmission speed from what they are to 19200.  That is most likely what's going wrong here. Whatever you changed, don't do that.  Change it back to defaults and try again.

highflux7, 2020-12-24:
Thanks for the quick response!
Yes, my fault. Reverted to transmission speed to 115200bps and can now successfully bootstrap via Send ProDOS followed by Send ADTPro Serial Client.

But there's something still not working or I *really* don't understand about the program.  With the ADTPro Serial Client running on the Apple, still connected to the ADTPro server program running on the linux machine, how now does one start transferring image files??  (say starting with the ADTPRO-2.1.0.DSK & ADTPRO-2.1.0.PO)  I've got a floppy in the drive, formatted it, set ADTPro to (R)eceive and it asks for a filename.  I hit return and it says "WAITING FOR HOST REPLY, ESC CANCELS".

On the linux server side, where does one select or initiate transfers of disk images??  In the 'File' menu, I can select the working directory and indicate files of all type, but no files are shown. The Bootstrapping menu has various options, but I don't see anything in there about transferring images other than the bootstrapping varieties.  I've tried Disconnecting and reconnecting, but that doesn't seem to make any difference either.  What am I missing??

Thanks again for your time!

david-schmidt, 2020-12-24:
So far, all I'm reading is that you're able to push data to the Apple II, but the Apple II isn't pushing anything back.  If Speediboot isn't working, that's also a sign - it uses two-way communications, so if something isn't right in your cabling or settings, Speediboot won't work.  At this point, your cabling seems the most likely culprit.

highflux7, 2020-12-24:
Oh, that makes complete sense.  I'll make a totally new cable and see if that doesn't solve the problem.
Kind regards.

highflux7, 2020-12-25:
OK.  So I made a new cable, but am getting the exact same results with it that I did with the previous cable.

Next, to verify that 2-way communication was indeed happening between the apple and the linux box, I ran ProTerm on the apple and minicom on the linux machine and found no problem whatsoever-- they were communicating fine, tx on one, rx on the other, etc. both directions.  Even transfer files via xmodem no problem.

Now what??  What more can I do to debug this situation with ADTPro?

david-schmidt, 2020-12-25:
There was a changeover in underlying serial comms software in the most recent version of ATDPro (2.1.0).  It might be interesting to try the other version (2.0.3) to see if you have any different luck.  You didn't mention which version you're running - but my suggestion is to try the one you haven't tried yet.

highflux7, 2020-12-25:
Yes, I was using v2.1.0.  Unfortunately it will be a bit before I can check the older version as the Apple //e broke while after I swapped a card in/out and am left with a dead machine at the moment.  Hopefully I will get it running again soon and will be able to give it another go.

highflux7, 2021-01-05:
Repaired the //e and ADTPro 2.1.0 is now working fine!  Speediboot now performs flawlessly.

david-schmidt, 2021-01-05:
Good for you - what ended up being the problem?

highflux7, 2021-01-06:
Long story short, the mmu chip died.  Scoped everything I could find and pretty much narrowed it down to the mmu.  Ended up replacing the motherboard as those mmu's are hard to come by.  Next discovered that either/both the bus tranciever and latch chips on the 64k 80-column board were flaky too; all the memory tested fine.. just the bus chips were dotty.  When it rains it pours...

closed 2021-01-06 by highflux7