ADTPro/adtpro #124: Cannot launch ADTPro Client

AngieAndretti, 2021-01-09:
I want to transfer a disk image from a Win98SE PC with a COM port to an Apple IIc+.  I have the requisite cable and I'm stuck on software step 1, launching the ADTPro client on the host PC.  I installed Java v.1.5.0_22 and "java -version" works at the command prompt.

However ADTPro.bat just produces a console window that disappears almost instantly.  I was able to slow down the PC enough to capture some of the output: Exception in thread "main" java.lang.UnsupportedClassVersionError: Bad version number in .class file...

So I thought, OK, maybe this just has no love for the platform I'm trying to run it on - so I installed Java on my Win10 laptop and the result is very similar.  Java appears momentarily in the taskbar but disappears with no output.  I need some guidance please.

AngieAndretti, 2021-01-09:
UPDATE: I got the client to launch on the Win10 PC using the very latest version of 64-bit java, so I assume the problem is the older version of java being used on the Win98SE PC?  Still that's a little strange because the ADTPro.bat mentions version 1.4.2 in the example path variable and I'm running 1.5.0 on this machine.  Is there any way I can get it running on this platform?

closed 2021-01-09 by AngieAndretti

david-schmidt, 2021-01-11:
Java 1.6 is the absolute minimum required these days.