ADTPro/adtpro #126: Crashes out when selecting "serial" Win-10

baldrick1, 2021-02-02:
I've gone though several issues that are now closed but none have had advice that worked.
I am using an FTDI USB-Serial adapter
Running Win-10 64 bit.
ADTPro sees the com port (3 in this case) but also sees whatever I set the com port to in the properties manager.

Same issue in v2.0.1 and 2.0.3
Error log is attached.

Thanks in advance for your help.
[hs_err_pid10984.log](https://github.com/ADTPro/adtpro/files/5908289/hs_err_pid10984.log)

david-schmidt, 2021-02-02:
Is there any particular reason why you're using the older version 2.0.3 and not the latest 2.1.0?

baldrick1, 2021-02-02:
Inertia, mostly.  This has worked properly in the past.  I'm suspicious of Microsoft's guerrilla updates, personally, but hoping you might have any ideas in the interim.  I' will install v 2.1.0 and report back today.

david-schmidt, 2021-02-02:
The guerrilla update that got you was Oracle's, not Microsoft's.  If you look through the closed issues, there are several having to do with the level of Java that broke the JNI layer to rxtx, which is the reason for the v2.1.0 release.

baldrick1, 2021-02-02:
Thanks for the tip.  I bow to you and your patience. 
Of course running the new version works perfectly on Win-10 now.

david-schmidt, 2021-02-22:
Sounds like things are back to normal now - closing.

closed 2021-02-22 by david-schmidt