ADTPro/adtpro #127: Host Timeout in Windows 10 no matter what

ataru67, 2021-02-14:
Hi there, I've been using ADTpro for years now, although primarily on an older Pentium 4 machine, which I no longer have. It was extremely reliable on that machine, only a few times would I have to redo a disk write, and I have lots and lots of fully functional disks to show for it. I use the USB-DB9/DB9-ADB cables, and the printer port on a IIgs (rom 01). Let it be known that 2.0.1 was the current version when I first started using it, so that's what I currently have a working disk for.

Now I am trying to start using it again to archive some disks for somebody but currently on a Windows 10 machine and it's simply not working. The computers are definitely communicating, I get status updates on the windows adtpro client of what's happening and the serial status comes up "connected". The baud rates are set the same on both systems (have tried both 92k and 115k) and I am constantly getting host timeout errors, and almost immediately. Sometimes I will get directory listings but when I select a file (right now I'm simply trying to make a disk from the current version .dsk file) it just times out again. Very frustrating.

Now I know I should probably be using the current version, hence why I'm trying to make a disk of the current version. But the fact that it worked fine on a much slower Windows 7 machine and isn't working reliably at all on a newer windows 10 machine has me stumped. Any ideas? I really need to archive these disks soon but can't at the moment...thanks!

david-schmidt, 2021-02-14:
There’s a couple of usual suspects.  It’s most likely your USB cable, or the driver you’ve got installed on Windows.  But one good reason to get the latest server software on your Windows box is that there’s a very recent update in Java interface code that will be important to you.

If you’re using the same USB adapter on the Windows 10 box, then it’s the driver based on your description so far. If it’s a different USB adapter than the one that was successful before, that’s the problem.

ataru67, 2021-02-14:
Today I deleted the USB-to-serial driver that windows automatically installed and found one specific to my cable to replace it with. I then used the "old" bootstrapping method so I was running the same version as the windows machine, both 2.1.0. The bootstrapping worked perfectly both times, not a single bit of corrupt data, so I'm assuming the cable is working perfectly.

However once I was in ADTpro again, I was getting the exact same errors, host timeout 90% of the time, occasionally displaying directory contents, usually not. I can't imagine the cable is the issue if the bootstrapping went flawlessly both times, but the bootstrapping occurs at a much lower baud rate so who knows. I'm totally stumped here.

david-schmidt, 2021-02-14:
If you used the old bootstrapping method, you’ve shown that one direction works reliably, but it doesn’t say anything about the other direction.  If you were to use Speediboot, it would test bi-directional communications. 

Also... earlier, you referenced 2.0.1; just now, you referenced 2.1.0.   They are very different things...

ataru67, 2021-02-14:
Yes, I've been using 2.1.0 on the windows machine from the start, but my disk copy of ADTpro was 2.0.1, which I figured was causing an issue. Since I can't use adtpro currently to make a physical 2.1.0 disk yet, I bootstrapped into 2.1.0 today to see if having both machines on the same version would solve my issues. Now both machines are running 2.1.0, but no success.

I did try speediboot, but it didn't seem to work. Since that sends data both directions, maybe my cable is to blame, although I'm not sure what would have failed between using my old machine and this new one. I'm going to try on my IIe with a super serial card on the off-chance my IIgs modem port has a broken solder joint or something. I'll report back once I've tried it on the 
IIe. I appreciate the help!

david-schmidt, 2021-02-14:
If Speediboot doesn’t work, it’s 99% likely your cable’s problem and 1% the port on the GS.  Roughly.

ataru67, 2021-02-14:
Well I just tried it on the IIe SSC setup using just straight USB-to-DB25 and the second I enter serial mode in ADTpro the screen goes blank and I get a constant noise out of the speaker. Not sure what that means, but I'm going to just order a different cable and see what I can get with that. Thanks for your patience. If a new cable doesn't help, I'll continue this thread.

david-schmidt, 2021-02-14:
Yeah, no, the cheap clone ones are trash.

ataru67, 2021-02-14:
Haha I assumed so. Guessing I'll be trouble free when the new one arrives. Thanks!

ataru67, 2021-02-21:
as you said, a new cable would fix it. Absolutely runs perfect now. Thanks for your patience and assistance!

david-schmidt, 2021-02-22:
Awesome, glad you got it working.

closed 2021-02-22 by david-schmidt