ADTPro/adtpro #128: Unable to transfer files (host to client) or poll directory on host

vastargazer, 2021-02-17:
- I am running ADTPro 2.0.3 on a MacBook Pro Mojave 10.14.6.  Using a SSC in a Apple IIe.  Set to straight through and on terminal block and DIP switches are SW1: 1001111  SW2: 1101100.  I was previously (a few months ago) able to create an ADT boot floppy but now unable to even repeat that process.

- I am using a Keyspan 19H as a USB to serial converter. Latest Keyspan drivers/software installed.  System sees the adapter as does ADTPro 2.0.3.

- The MacBook is able to bootstrap and load the ADT Serial Client and I can start ADTPro from the previously created floppy.

However, once the ADT client starts, I am unable to retrieve directories or receive files.  The error is a Host Timeout.  It appears as if host to client communication ceases.

As an aside, I am unable to use 2.1.0.  It will not display the available serial ports.  

Any assistance is most appreciated.

Cheers,
mark

vastargazer, 2021-02-18:
Part of problem solved.  I have 2 SSCs in my Apple IIe.  Once the bootstrap transfers, the message then tells me Found Slot 1.  I use the Slot 2 SSC for ADTPro and Slot 1 for my ImageWriter.  All DIP switches are set properly per the respective manuals.

Once the bootstrap transfers using speediboot, the message then tells me ---> Found Slot 1. But the bootstrap was received properly by the Apple.  This occurs when the client is attempting to transfer from the server.  

Both cards need the block setting to Terminal.  

It appears the server will send the bootstrap to the client but then the client gets confused about which slot the ADTPro card is actually installed.

After running out of ideas I decided to remove the printer SSC and everything worked fine.  So, the question now becomes how do I get two SSCs to live happy with one another allowing both the ADTPro configured card and the printer configured SSC to live happily together.

2.1.0 still doesn't see my KeySpan USB to serial.  :-(

david-schmidt, 2021-02-18:
Two serial cards is a problem on bare metal bootstrapping because you don't get to choose which one to use after the first couple of several steps is taken.  You can IN#2 so on, but if you also have a valid serial card in slot 1, the automatic serial sniffing code is going to find it and use it first.  So you can start yourself in slot 2, but once the serial bootstrapper starts running, you don't have an opportunity to choose among multiple cards.

The solution is to do just as you did: pull out the serial card in slot 1 (or use it all the way through - tougher if it's set up as a printer interface), make yourself an ADTPro boot disk, and then save the client configuration on that disk so you always tell it to use slot 2 even though another is available.  I know it's great fun to do bare metal bootstrapping, but you should only ever have to do that once.

It's disappointing to hear about the Keyspan adapter not showing up - that's a failure deep in the Java serial library, it sounds like.  It's the thing that enumerates what is available on the hardware side of things.  It sounds like the older (rxtx library) is able to see it, but the newer (jssc library) is not.  That's the functional difference between 2.0.3 and 2.1.0.  Which is a shame, because the KeySpan/TrippLite adapter was once the Cadillac of adapters, especially on the Mac.  I have to say I don't have a ready solution for that beyond seeking out an FTDI-based adapter.  I'd be happy to trade you one with you if you're interested to be able to research if there's anything that can be done on our end.

vastargazer, 2021-02-18:
Hi David:
Thanks for the excellent suggestion.  
I have a 2.0.3 boot disk.  I never thought about saving the configuration to the ADTPro boot disk.  I will definitely give this a try.
When this all started, both cards were in the Apple IIe, and the same host time out occurred when I booted from my ADTPro boot disk.  My initial instinct was to blame a cabling errors.  Checked cables with multi-meters to ensure I was using a straight through, etc.  Before posting here, I was using the bare metal boot to try and confirm bi-directional communication between host and client since receiving was timing out, even with the boot disk.  (Somewhere I read speediboot was a good way to confirm that). 

But, I will definitely try saving the configuration and see if that solves the problem.

With respect to the RXTX issue.  That makes total sense.  After I saw the error message about that library missing, I started doing so general research on the that particular aspect of Java.  If I get anywhere with it I will be certain to share my findings here.  I have some colleagues who are Java gurus, perhaps they can point us in a direction, with respect to Macs, as I get the impression that 2.1.0 works properly on Windows platforms.

Thanks again for the help and for creating this awesome tool for the vintage computer community.

Cheers,

mark