ADTPro/adtpro #129: Apple III Serial card Slot choices 820-0071

Verault, 2021-02-25:
So I have some Apple Brand 820-0071 Serial adapters. From everything I have read they are identical to the onboard Serial of the Apple III Mainboard. So the way I see it, can we add the option to the client for ADTPro for the Apple III to choose which slot the Card is in? That way we no longer just have the choice for the onboard serial adapter, but also another Apple Serial adapter in whichever slot we choose?

Verault, 2021-02-25:
Here are photos of the serial board: https://photos.app.goo.gl/bv2umN8mhRWa3PdP9

david-schmidt, 2021-02-25:
Today, you can choose SSC cards to put in any of the 4 open slots:
[image]
If the 820-0071 cards are in fact the same hardware as the main board, then the hardware addresses of the ACIAs are going to differ from SSCs.  So I guess an option would be to cycle through cards and slots:
SSC/1, 820/1, SSC/2, 820/2, SSC/3, 820/3, SSC/4, 820/4

Verault, 2021-02-25:
I think that would be a perfect selection to choose through. Would that be a relatively easy implementation?

david-schmidt, 2021-02-25:
It would be straightforward, and there's headroom for code on the /// that isn't as easy to come by on the II.  If you're not volunteering to propose a code change, I'd prefer to have real hardware in hand to test myself.

Verault, 2021-02-25:
I'd be happy to send it to you as a loaner. Just as long as I can have it sent back after final testing.

Would that work? Ill pay shipping fees of course.

Verault, 2021-03-12:
David, just contact me with where you would like me to ship this card. Thank you.