ADTPro/adtpro #130: Latest version v2.1.0 crashes on some Raspberry Pi models

quorten, 2021-03-07:
I've observed for ADTPro 2.1.0, after about two seconds from starting, it crashes with "illegal instruction" on a Raspberry Pi Zero running Raspbian 8 (jessie), but v2.0.3 start fine.  However, ADPro 2.1.0 starts up just fine on a Raspberry Pi 4 with Raspbian 10 (buster) 32-bit.  The crash backtrace appears to point into the serial library:

```
Stack: [0xb62b3000,0xb6303000],  sp=0xb63001e0,  free space=308k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libjssc.so+0x3b68]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  java.lang.ClassLoader$NativeLibrary.load(Ljava/lang/String;Z)V+0
j  java.lang.ClassLoader.loadLibrary0(Ljava/lang/Class;Ljava/io/File;)Z+328
j  java.lang.ClassLoader.loadLibrary(Ljava/lang/Class;Ljava/lang/String;Z)V+48
j  java.lang.Runtime.load0(Ljava/lang/Class;Ljava/lang/String;)V+57
j  java.lang.System.load(Ljava/lang/String;)V+7
j  org.scijava.nativelib.NativeLibraryUtil.loadNativeLibrary(Lorg/scijava/nativelib/JniExtractor;Ljava/lang/String;[Ljava/lang/String;)Z+129
j  org.scijava.nativelib.NativeLoader.loadLibrary(Ljava/lang/String;[Ljava/lang/String;)V+13
j  jssc.SerialNativeInterface.<clinit>()V+86
v  ~StubRoutines::call_stub
j  jssc.SerialPortList.<clinit>()V+0
v  ~StubRoutines::call_stub
j  org.adtpro.transport.SerialTransport.getPortNames()[Ljava/lang/String;+8
j  org.adtpro.gui.SerialConfig.enumeratePorts()V+30
j  org.adtpro.gui.SerialConfig.<init>()V+242
j  org.adtpro.gui.SerialConfig.allocateSingleton(Lorg/adtpro/gui/Gui;)V+11
j  org.adtpro.gui.SerialConfig.getSingleton(Lorg/adtpro/gui/Gui;)Lorg/adtpro/gui/SerialConfig;+12
j  org.adtpro.gui.Gui.<init>([Ljava/lang/String;)V+1797
j  org.adtpro.ADTPro.main([Ljava/lang/String;)V+51
v  ~StubRoutines::call_stub
```

Attempting to disable JIT with `-Xint` made no difference.

Probably worth investigating this issue more to see if the serial library change introduced an ARM CPU revision dependency or it's just due to the old version of Raspbian.

david-schmidt, 2021-03-08:
Makes sense, the change between 2.0.3 and 2.1.0 was the serial library that talks to the hardware.  So that's where the problem is going to lie... somewhere between the serial library (https://github.com/java-native/jssc) and the hardware it's talking to.  I'll leave this open so others can see it, but it's not something ADTPro can do anything about until jssc addresses it.

ben-wing, 2021-05-05:
I would expect ARM CPU revision is the difference .
I'm running Raspbian 10 (buster) on Raspberry Pi Zero, and am seeing the same behavior as @quorten, so the issue is not with their Pi Zero running 8 (jessie).
However the Pi Zero is an ARMv6, and Pi 2/3/4 all are ARMv8

I've planned to recompile jssc on the Pi Zero in case natives/linux_arm/libjssc.so isn't compatible with the ARMv6

iKarith, 2021-08-19:
That's pretty much a guarantee if whomever built the "armhf" version was doing so on a Debian system as Debian presumes ARMv7 for "armhf" architecture. The little Pis are inconveniently ARMv6 which in Debianland was armel, ARMv3, no hardfloat. Old Raspbian versions used armel, but were hamstrung by this. Unfortunately, they choose to redefine armhf rather than define a new architecture, which creates problems like this every now and then.

Worse, I see the package is not simply in the Raspbian package pool all ready to go, so there's likely an additional problem.

timriker, 2024-05-09:
Has anyone found a work around for this?
Is the apple 2.1.0 client completely compatible with the 2.0.3 server?

david-schmidt, 2024-05-10:
1. Not so far as I'm aware.
2. Yes.