ADTPro/adtpro #131: Audio Disk image transfers

mkubiak402, 2021-05-12:
I am able to get ADTPro booted up with the Audio option. I have a disk formatted but I Can't make the disk image I type in "ADTPRO-2.1.0.DSK"  or "ADTPRO-V.R.M.DSK" and the apple IIe just sits there with "waiting for host reply"

if I set the PC to Listen to "line in" I can hear a little tone from the apple IIe but its faint so I think the problem is the sound level is just to low for ADTPro to hear it? do I need to run some kind of amp?

david-schmidt, 2021-05-12:
Here's everything I know about that:
https://adtpro.com/connectionsaudio.html#Your_Microphone_May_Need_Boosting

mkubiak402, 2021-05-21:
Nope I've tried a myriad of settings my headset works fine the Apple just does not put out enough signal for my computer to hear I'm going to try using some kind of amplifier other than that there's no way for me to turn it up anymore..
Thanks any how.

ryandesign, 2021-09-21:
I too saw this problem when using the built-in audio of a 2011 MacBook Pro. With the input volume all the way up, the volume meter in the Sound preference pane still showed a signal of only about 25% of maximum volume coming in. Switching to a USB audio interface resulted in an incoming audio signal of about 50% of maximum and this seemed to work.