ADTPro/adtpro #133: Aplle II gs ROM 1

syncpak, 2021-06-25:
When I try booting my Apple IIgs / ROM 1 with ADTPro 2.10 i can get to the menu where you can choose serial, ethernet, audio. When i choose the Serial option, it goes into a black screen and hangs
When i boot with the older ADTPro 2.03, and I choose the Serial option it works ok and goes to the next screen.
Is there something that I can tweak that could give you more information ?
Thanks

david-schmidt, 2021-06-25:
Do you have a serial cable (null modem) plugged in to port 2?  Does 2.0.3 communicate correctly?

syncpak, 2021-06-28:
Yes, Its the correct cable (null modem) on port 2 and 2.0.3 communicates perfect. 
I have been transferring files with 2.03 and its works great.

david-schmidt, 2021-06-28:
Interesting.  What level of code is your server at?  Have you tried both 2.0.3 and 2.1.0 at the server end?