ADTPro/adtpro #134: What causes this? (Apple IIe --> Audio --> Win10)

rhubarbo, 2021-07-12:
When copying files out, I get this problem. Thanks for any information. 
[image]

david-schmidt, 2021-07-13:
That's pretty unusual, and the crash location ($6424) isn't in the normal code's memory space.  So, some typical initial questions apply:
1. Does this behave the same way on every disk that you try to transfer?
2. Has it ever been successful before on this machine - with any version of ADTPro?
3. Do you have any memory or system diagnostics you can run on the machine?

rhubarbo, 2021-07-13:
Thanks for your response: 

1. Does this behave the same way on every disk that you try to transfer?

- Yes, I tried it on the ADTpro disk itself. 

2. Has it ever been successful before on this machine - with any version of ADTPro?

- This is my first attempt at this so I have no previous attempts. 

3. Do you have any memory or system diagnostics you can run on the machine?

- I do not but will check this out and reply later. Could it also potentially be a problem with the drive or should I be looking at the internal hardware?

david-schmidt, 2021-07-13:
It's possible the bootstrapping inserted some errors during initial transfer that corrupted some memory, or it could be a hardware fault.  Do you have a boot disk, or are you doing audio bootstrapping?

rhubarbo, 2021-07-13:
I am using the ADTPro boot disk.

david-schmidt, 2021-07-13:
Great, that eliminates bootstrapping as a possible source of error.  It’s sounding a lot like hardware at this point.

rhubarbo, 2021-07-13:
Hmm. Thanks again - Are you thinking Floppy Drive or a chip somewhere?

david-schmidt, 2021-07-14:
It's not likely the drive, if it's able to boot ProDOS and load up ADTPro.  Do you see similar problems if you _receive_ a disk?

rhubarbo, 2021-07-14:
I will check that out tomorrow.

WilliamFulk, 2021-12-03:
Getting the same ADTPro 2.1.0 Audio (S)end image Image transfer aborted failure on my Apple IIe to HP workstation p6140f.  Error Msg on Apple IIe:  "6424-  A=7C X=0A Y=79 P=34 S=F4________"  (Note: made multi PC Microphone setting changes same failure)
The ADTPro Audio (R)eceive is working fine with PC headphone levels 20 left/right, Enhancements disables, Advanced 2ch-16bit-4800Hz and PC Microphone levels 92 +30db, Enhancements disables, Advanced 2ch-16bit-48000 .
Bought a USB Sound Buster Play! 3, ADTPro Audio (R)eceive worked fine but ADTPro Audio (S)end fails the same way. 
Is there a known solution to this issue?

rhubarbo, 2021-12-03:
I ended up buying a serial card and it blasted through without a hitch with a serial to USB cable. 

DSluga

On Dec 3, 2021, at 2:40 PM, WilliamFulk ***@***.*** ***@***.***> > wrote:

Getting the same ADTPro 2.1.0 Audio (S)end image Image transfer aborted failure on my Apple IIe to HP workstation p6140f. Error Msg on Apple IIe: "6424- A=7C X=0A Y=79 P=34 S=F4________" (Note: made multi PC Microphone setting changes same failure)
The ADTPro Audio (R)eceive is working fine with PC headphone levels 20 left/right, Enhancements disables, Advanced 2ch-16bit-4800Hz and PC Microphone levels 92 +30db, Enhancements disables, Advanced 2ch-16bit-48000 .
Bought a USB Sound Buster Play! 3, ADTPro Audio (R)eceive worked fine but ADTPro Audio (S)end fails the same way.
Is there a known solution to this issue?

renshordijk, 2021-12-16:
I'm having the same problem on my Apple II+ clone. I did found a way to reproduce the 6424 error. Just using ADTPro as a disk to send. All ADTPro values are default. Here are the steps to reproduce (I'm using a MacBook pro, on Windows it will be similar):
- On Mac: System preferences -> Sound -> Input tab (here you have to watch the incoming sound when the first 4 blocks are sent. You will notice by a continuous input level)
- Have the ADTPro Mac app running on audio as you normally would
- Start sending the ADTPro disk itself, when the first four blocks are sent over audio, quickly press "Mute" in the system preferences sound window for the "Output volume"
- When ADTPro Mac app says "Image transfer aborted" unmute sound again
- Just wait a few seconds and you will get the 6424 error

I wonder if the actual problem comes from some timings are being a bit off? This problem usually occurs for me after a few blocks that were successfully sent. When I send one block at once, the problem happens much quicker. Receiving disks on the other hand is totally fine.

I did run the XPS Cillin memory test for a few hours and no problems there by the way.

MGerckens, 2023-04-08:
[image]
Same exact problem here. Receving works fine. After around 4-6 successful blocks, it stalls for a bit and then does this. It's occasionally shown a different memory address or crashed entirely: 
[image]
Server says "Image transfer aborted". When I change blocks-at-once, it stops after 4-6 sets of blocks.

mheilemann, 2024-05-13:
I have been trying for a while and have the same exact problems.  Any other advice before I give up on sending disks from the IIe over audio?  I have booted from actors 2.1.0 disks, tried all the audio settings, and it happens intermittently, but never sends a full disk.  I have gotten about half way when using 5  blocks.

MGerckens, 2024-05-13:
I ended up reverting to 2.0.3 and it worked. I was using windows 7 though, I assumed it was a compatibility issue. I doubt that would be relevant on 10.