ADTPro/adtpro #135: Speediboot always works, everything else always Host Timeout

mozzwald, 2021-07-17:
I just acquired my first IIe and have issues with ADTPro. I first tried loading disks over audio from my android phone and from my computer which both failed, then I switched to using serial. The issue is that I can bootstrap Speediboot and get ADTPro client running on the Apple just fine, then 99% of the time I get Host Timeout errors. I have confirmed SSC terminal block and dip switch settings according to the guide.

I have 2 Targus USB port replicators that I tried first (MCT-232 serial) using some old serial cables and adapters. One of the Targus would **rarely** give me the directory listing or transfer a couple blocks before failing. The other would never work (although bootstrap worked fine). Today I went to the local computer shop and bought a "UGREEN" db9 usb to serial cable with FTDI chip and a db25 adapter which bootstraps fine and host timeouts everything else. 

I have tried with a Linux Mint Dell laptop, a Macbook Air dual booting 10.13 and Linux Mint and a Lenovo Win10 Laptop. All produce the same results. Here is a snippet of failure:

```7/16/21, 6:39:49 PM SerialTransport.pushBuffer() exit.
7/16/21, 6:39:49 PM CommsThread.sendPacketWide() calculated CRC: 19231
7/16/21, 6:39:49 PM CommsThread.pullEnvelopeWide() entry.
7/16/21, 6:39:57 PM CommsThread.pullEnvelopeWide() aborting due to timeout.
7/16/21, 6:39:57 PM CommsThread.pullEnvelopeWide() exit.
7/16/21, 6:39:57 PM CommsThread.sendPacketWide() Well, ack envelope was unexpected: 00
7/16/21, 6:39:57 PM CommsThread.sendPacketWide() ACK from client: 15
7/16/21, 6:39:57 PM CommsThread.sendPacketWide() didn't work; will retry #2.
7/16/21, 6:39:57 PM CommsThread.sendPacketWide() block: 0.
7/16/21, 6:39:57 PM CommsThread.sendPacketWide() backoff sleeping for 0 seconds (or 1 second, whichever is shorter).
7/16/21, 6:39:58 PM CommsThread.sendPacketWide() looping until successful to send block: 0000
7/16/21, 6:39:58 PM SerialTransport.writeBytes() adding 1 bytes to the stream.
... snip ...
7/16/21, 6:39:58 PM SerialTransport.writeBytes() adding 1 bytes to the stream.
7/16/21, 6:39:58 PM SerialTransport.pushBuffer() entry, pushing 235 bytes.
7/16/21, 6:39:58 PM SerialTransport.pushBuffer() pushing data:
C1 00 04 D3 16 00 00 44 05 09 F3 FE 11 FB 03 07 C7 2F F7 F4 E6 0F 39 07 FE F8 CA 3E 02 0B 00 18 
FD EA 0B F8 CB 15 2B 08 F7 FE 03 F2 03 0F BC 12 03 10 FC 22 FD BE 05 FC 03 FD 02 FF 35 05 0A F8 
08 BC DE 20 00 63 14 03 10 FC 02 FD DE 05 FC 03 FD 02 FE 16 0F F8 C2 34 03 10 FC 02 FD DE 05 FC 
03 FD 02 FE 22 FF BE 34 03 10 FC 02 FD 04 FC 04 DA 05 FC 03 FD 02 FE 16 0F F8 C2 3E E3 36 0F F8 
A2 42 23 F3 02 08 FF B2 34 1E EF 08 03 B4 21 ED 36 0F F8 A2 42 23 F3 02 08 FF B2 34 1E EF 08 03 
B4 22 EC 36 0F F8 A2 43 22 FD D5 0B 04 0C D3 2D D5 2B D3 FC 36 0F F8 A2 49 EE 0E F7 0D EF E8 05 
FC 03 FD 02 FE 16 0F F8 C2 49 13 09 02 01 EC 0B C2 42 FF 9E 49 13 09 02 01 EC 0B C6 FC 42 FF 9E 
F3 00 00 00 00 00 00 00 00 1F 4B 
7/16/21, 6:39:58 PM SerialTransport.pushBuffer() exit.
7/16/21, 6:39:58 PM CommsThread.sendPacketWide() calculated CRC: 19231
7/16/21, 6:39:58 PM CommsThread.pullEnvelopeWide() entry.
7/16/21, 6:40:06 PM CommsThread.pullEnvelopeWide() aborting due to timeout.
7/16/21, 6:40:06 PM CommsThread.pullEnvelopeWide() exit.
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() Well, ack envelope was unexpected: 00
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() ACK from client: 15
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() didn't work; will retry #3.
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() block: 0.
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() backoff sleeping for 0 seconds (or 1 second, whichever is shorter).
7/16/21, 6:40:06 PM CommsThread.sendPacketWide() exit, rc = false
7/16/21, 6:40:06 PM Gui.getWorkingDirectory(): /home/mozzwald/Downloads/ADTPro-2.1.0/disks/
7/16/21, 6:40:06 PM CommsThread.dispatchCommand() exit.
7/16/21, 6:40:06 PM CommsThread.commandLoop() Waiting for command from Apple.```

I also tried using the older 2.0.3 version with the same results. Any suggestions for something else to try?

david-schmidt, 2021-07-17:
Rule #1: it’s your serial cable. 
Rule #2: see rule #1.

So far, it sounds like your USB adapters (and their drivers) are probably ok, but you haven’t said anything particularly helpful about “some old serial cables and adapters.”

mozzwald, 2021-07-17:
> Rule #1: it’s your serial cable.
> Rule #2: see rule #1.
> 
> So far, it sounds like your USB adapters (and their drivers) are probably ok, but you haven’t said anything particularly helpful about “some old serial cables and adapters.”

I'm assuming the links from your reply are by accident and automatically pulled in as markdown from the github parser. They seem unrelated to the issue I'm having.

I've read countless responses about it always being the cable which is why I went out and got a [brand new usb to DB9 serial cable](https://www.amazon.com/dp/B00QUZY4L0) and a new DB9 to DB25 adapter as I mentioned before. I didn't mention the original cable(s) I tried because I figured the [brand new FTDI](https://www.amazon.com/dp/B00QUZY4L0) would rule out the serial cable/adapter as being the problem. All the cables I have are straight through. I have various combinations of male/female db9/db25 cables and adapters. I used the combinations to get from DB9 on the USB Port Replicator to DB25 on the IIe. What would you like to know about them?

mozzwald, 2021-07-20:
Made a little discovery. I can do two way serial comms with a terminal at 9600 baud but if I jump up to 19200 or higher then I get garbage. There may be a problem with the SSC. Any chance you can make a version with lower baud rates?

david-schmidt, 2021-07-20:
I took out the lower baud rates because it created more opportunities for failure (by mismatching baud rates), and if it feels like it becomes necessary to slow down, something else is wrong.

You say speediboot works - but after the initial textual dump, it shifts into 115.2k to load everything else (and is bi-directional).  So by all measures, this should be working.  Can you go into detail about the Apple II you've got, the slot the SSC is in, and the pinout of your DB25 adapter?

mozzwald, 2021-07-21:
> I took out the lower baud rates because it created more opportunities for failure (by mismatching baud rates)

If a user is changing baud rates I would expect they could do it on both ends. Having more than 1 baud rate option creates this 'opportunity for failure' so adding even more should not be a problem in this case since you already have 2 options. 

> if it feels like it becomes necessary to slow down, something else is wrong

agreed, but if I can use something (partly broken) at a slower speed and it works, I'm willing to wait for it to transfer at a lower speed. If my car overheats at 50mph, but will run fine at 30mph, I'll take the slow route ;)

> You say speediboot works - but after the initial textual dump, it shifts into 115.2k to load everything else (and is bi-directional). So by all measures, this should be working. Can you go into detail about the Apple II you've got, the slot the SSC is in, and the pinout of your DB25 adapter?

I just got this Enhanced IIe with AppleColor Monitor 100 and duodisk drive in an auction. It has a SSC card in slot 2 currently and tried in a few other slots which had the same problem. I have [this DB25 adapter](https://www.ebay.com/itm/154379318441) and this [USB/Serial adapter](https://www.amazon.com/dp/B00QUZY4L0) that I purchased locally. FWIW, I also tried with a tower PC running Linux and a real serial port which had the same problem.

I toned out the D9 to D25 adapter and this is what I got:
D9 -> D25
DCD 1 -> 8
RXD 2 -> 3
TXD 3 -> 2
DTR 4 -> 20
GND 5 -> 7
DSR 6 -> 6
RTS 7 -> 4
CTS 8 -> 5
RI 9 -> 22
..which makes it a straight through adapter. I have my terminal block on the SSC set to terminal and the jumpers set according to the ADTPro docs:
[image]

david-schmidt, 2021-07-21:
I don't see anything amiss in your configuration.  One question remains unanswered: you say speediboot works.  That is 115.2kbps, and it is bi-directional.  That's at odds with subsequent comms failing.

So, can you run a speediboot from bare metal, and once the ADTPro client comes up, do a `G` and report out what it's saying for the comms port it finds, and speed it's set to?  The code to sniff the port is identical in speediboot vs. ADTPro, but given the incongruity of your situation, some assumption is not going the way we expect...

mozzwald, 2021-07-24:
Today, I acquired another SSC card which was a working pull from another system. I tested it with ADTPro and have the same problem. The client appears to detect the card and 115200 baud:
[image]

Still, anytime I try getting a (D)ir listing the client shows _HOST TIMEOUT_.

mozzwald, 2021-08-04:
and I made a video of the failure https://youtu.be/a3cNEJN66Q8

david-schmidt, 2021-08-04:
Cool, looks like the host hears something - you can see the status bar change a little bit when you send in a command.  A trace might show more of what the symptoms are, but now the variable in question is the cabling.

mozzwald, 2021-08-04:
> A trace might show more of what the symptoms are

I posted a complete successful bootstrap, failed (D)ir listing, successful (C)D and failed (D)ir listing at https://gist.github.com/mozzwald/4b8875b5f8adf8bca292f33a2ec9c61a

david-schmidt, 2021-08-04:
It looks like that got chopped off in the middle.  Can you just capture a (D)ir scenario?  Delete the trace file, start tracing, do the (D)ir, stop tracing, and send that.

mozzwald, 2021-08-07:
Turns out this IIe has a bad RAM chip so I'm closing the issue.

I acquired 2 Apple II+ units this week and was able to create a floppy with those using the same USB FTDI adapter and serial connectors I was using on the IIe. So, it's *not* always the cable ;) My lack of Apple knowledge did not help. After much searching I found out the IIe had a builtin self test, go figure.

Thanks for working with me on this!

closed 2021-08-07 by mozzwald

david-schmidt, 2021-08-07:
Fantastic.  Two observations:

1. Hardware problems are exceedingly rare - but not unheard of.  Definitely not the first thing to suspect.  (When you hear hoofbeats, think horses, not zebras)
2. Given the incongruity of the symptoms (two things that simply didn't go together: successful 2-way comms with speediboot vs. nothing else) failing hardware does make sense.

Glad you got it working!