ADTPro/adtpro #138: .WOZ format disk images

ryandesign, 2021-09-22:
The ADTPro web page says it:

> Understands many digital disk image formats: .DSK, .SDK, .SHK, .BXY, .PO, .DO, .NIB, 2IMG

Does it understand the [.WOZ disk image format](https://applesaucefdc.com/woz/)? If so, it should indicate that. If not, can support for that be added?

david-schmidt, 2021-09-22:
The .woz disk image format is all about capturing the minutiae of copy protection schemes, none of which a standard disk controller can reproduce.  In the degenerate case where there is only standard in-band track data, the .woz wrapper is wholly unnecessary.  So the intersection of it and ADTPro is the null set.

closed 2021-09-22 by david-schmidt

ryandesign, 2021-09-23:
I see. I was using the [Virtual \]\[](https://www.virtualii.com) emulator to try to produce some disk images that I could transfer to my real Apple //e. I had created the disk images in .dsk format to start with, but after making changes to the disk (using Ultra Fast Pix which writes the entire track as a single sector for speed) and ejecting the disk, Virtual ][ said the disk had been changed in a way that could not be represented in its current format, and it suggested that I use the .woz format instead. But if ADTPro doesn't and won't support that format, then that won't meet my needs either. The third format Virtual ][ supports is .po, so I guess I'll try that next.

ryandesign, 2021-09-23:
Virtual ][ can only create disk images in .dsk or .woz (version 1 or 2) formats. AppleCommander can create images in DOS order (.dsk) or ProDOS order or nibble order (.po). A nibble-order .po cannot be inserted into Virtual ][ (it says the disk has too many tracks). A ProDOS-order .po can be inserted but when I save data to it with Ultra Fast Pix Virtual ][ again refuses to save the changes when the disk image is ejected.

Seems like I'm reduced to transferring the images to the real Apple II on a normal DOS 3.3 disk and using the Apple II to save the images to another disk with Ultra Fast Pix, which I was hoping to avoid because it will involve a lot of disk swapping because I am having a problem with one of my Disk ][ drives.

david-schmidt, 2021-09-23:
There are some situations where you will simply not be capable of re-creating a disk given the hardware you have.  Disk imaging is not always a 2-way street.  I'm not familiar with Ultra Fast Pix, and I can't seem to locate it online to look at what it's doing - but it also seems unlikely that a retail-level program is going to produce copy-protected disks, so I have a feeling that Virtual ][ is just mistaken in what it thinks is happening at the drive controller.  If you can point to somewhere I can download this program, I might be able to take a look.

ryandesign, 2021-09-23:
> There are some situations where you will simply not be capable of re-creating a disk given the hardware you have. Disk imaging is not always a 2-way street.

As I'm discovering! I certainly understand your point that ADTPro could not be enhanced to transfer copy protected media that relies on qualities a disk acquires during commercial duplication that cannot be replicated by a consumer disk drive.

> Ok, found it:
> https://www.applefritter.com/appleii-box/G74UltraFastPix.htm

That's the one. Apologies; I should have provided the link. It's not a commercial program, just a library and an example of how to use it to store 17 HGR pictures on a disk and read them back in, forgoing the disk catalog, filenames, and smaller sector sizes in the interest of loading the pictures as fast as possible just by specifying the picture number 0-16.

The .dsk linked at the bottom of that page doesn't actually contain Ultra Fast Pix; it contains its predecessor, Fast Pix. Since it took me awhile to find, I'll mention that Ultra Fast Pix was in the March 1987 issue of Nibble magazine whose programs are contained on disk 30 of [Nibble Programs.zip](https://web.archive.org/web/20200101192830/http://nibblemagazine.com/Nibble%20Programs.zip)

> So while a physical Apple would be able to reproduce a disk like this from disk to disk, there's not a way to represent what it's doing without a .woz wrapper.

Does that change your mind about the usefulness of adding support for at least some aspects of the .woz format to ADTPro? Or is it the case that ADTPro can only transfer disks that use well-known formats and that it would not be capable of transferring arbitrarily-formatted disks like this one? If the .woz format stores the raw bits/bytes that go on each track, can't Java ADTPro send that information to Apple II ADTPro which could then write those bits/bytes back onto the tracks? I don't know ADTPro's internals so apologies if I'm thinking about this too simplistically.

The method Ultra Fast Pix uses results in reads that are five times faster than DOS 3.3 and allows 17 HGR pictures (with one track left over) to be stored on a disk where only 16 could be stored with DOS 3.3. These speedups and space savings would certainly have been of interest to developers at the time so I would assume some software was distributed on disks that made use of these techniques so it could be of benefit for ADTPro to be able to transfer these kinds of disks.