ADTPro/adtpro #141: All @ symbols when bootstrapping

michellybells, 2021-09-28:
I recently got my first Apple IIe machine complete with a Super Serial Card (in Slot 2) and two floppy drives, and I plan to use ADTPro with it for disk images. I also have an SDISK II LCD flash device so that I can load images, but of course I prefer real floppies. I have two serial devices connected to a Windows 10 machine. One is a USB interface using the CP2102 chipset (not the best, I know) and I also have a PCI serial card with a 9 to 25 pin cable.

I first tried using the ADTPro boot disk as an image to connect in serial mode with the USB interface. I set the dip switches as recommended on the guide, and the jumper block is in Modem mode (pointing up). In this situation, the Apple II can load serial mode and recognizes the connection with ADTPro. However, any attempt to transfer data ends immediately with a host timeout message.

I then started testing without the disk to see if I could bootstrap. Whenever I attempt to bootstrap, at either 9600 or 2400 (set both in ADTPro and the Apple II) instead of hex data I get nothing but @ symbols with / at certain stop points. After that, I tested it with the PCI interface and found the exact same results. I checked all the Windows 10 COM settings for both ports, and tested a variety of different jumper settings on the SSC. Putting the jumper block in Terminal mode makes no connection at all, so I have since left it in Modem mode.

On the advice of some friends, I next attempted to connect the Apple II to a terminal program (PuTTY) with similar results. No matter what baud rate I tried, any text written on the Apple printed into the terminal as what looked to be ASCII 176 or 177 ▒ and when I type in the terminal the Apple prints as @ symbols. After a ton of experimenting with various settings, I found that if I select 7 data bits and MARK parity, I can get both the Apple and the terminal to communicate with no problem. (I believe even parity also worked, as long as it was 7 bits.) Since ADTPro only uses the 8 data bit mode, I can't test the software with 7 bit to see if anything changes. Note that 7 data bits works with both the USB and PCI interfaces.

My next step was to test ADTPro on a Macbook using the USB interface, and the results were the exact same.

Either I'm doing something really weirdly wrong, or there's something wrong with this SSC card. Any ideas?

david-schmidt, 2021-09-28:
You've done a great job of proving there's only one component that's failing.  Unfortunately, that component is the SSC.

michellybells, 2021-10-01:
I got lucky and found someone here in Japan selling a complete in box SSC at a pretty decent price, so I bought it. Sure enough, it worked perfectly with ADTPro first time! Guess eventually I'll have to repair the other SSC so I can sell it or use it as a printer port maybe. Is there any particular component of an SSC that is more likely to fail or known to cause this sort of problem?

david-schmidt, 2021-10-01:
That is lucky indeed.  Unfortunately (again), the most common failure on that card is the 6551 ACIA, and you will need similar luck in finding a working specimen to replace it with.

closed 2021-10-01 by david-schmidt