ADTPro/adtpro #143: ADTPro 2.1.0 not launching on Apple IIgs

dcgibbons, 2021-11-14:
This is the first time I'm using ADTPro at all on this hardware. I'm using brand-new cables (Apple IIgs to RS232 null on one side, and another with RS232 to USB on the other) connected to my macOS systems (I've tried both an M1 mac mini and a x86 2018 MacBook  Pro).

I'm able to get the IIgs booted and receiving data no problem. But at the end of the transfer, execution does not start. This doesn't seem to matter if it's ProDos or the simple client.

I tried changing the Pacing setting all the way down to 1000 just to make sure that wasn't an issue - no change in behavior.

I've also removed all of the cards from the IIgs to makes sure it's completely stock.

User error or is something else amiss here?

[image]

david-schmidt, 2021-11-15:
It looks like you're doing the manual bootstrap method, and it's not looking good.  Is there a reason why you're not doing Speediboot instead?  What does that look like?

dcgibbons, 2021-11-15:
Speediboot was doing the same thing, actually.

All that said, today everything is working. The only thing I think I changed was switching Slot 4 from user card to the default in the IIgs' control panel. I will play with this more and see if I can recreate again with some consistency and re-open if needed!

closed 2021-11-15 by dcgibbons