ADTPro/adtpro #145: ADTPro will only load Prodos on //c

PeaceFrogger, 2022-01-17:
I'm using windows 7 64bit over serial to a //c. When using Speediboot it loads ProDOS. When trying manually it loads ProDOS but can't get it to load ADTPro Serial Client

david-schmidt, 2022-01-17:
What happens after Speediboot loads ProDOS?  Can you shoot a video of what you're doing?

PeaceFrogger, 2022-01-17:
[image]
Now it won't load past this screen 
[image]

PeaceFrogger, 2022-01-17:
Ok. Now it loaded into ProDOS
[image]
[image]

PeaceFrogger, 2022-01-17:
When I type in in#2 and enter I get a beep. Can't seen to upload a video

PeaceFrogger, 2022-01-17:
I must be having a problem with my //c because the characters are not right

david-schmidt, 2022-01-17:
That sounds like a likely scenario from what you've described so far.

PeaceFrogger, 2022-01-17:
I was thinking I might have a ram problem. I will have to crack it open and find out. Thank you