ADTPro/adtpro #146: Can't access files

Drakepirate, 2022-02-10:
Hi,
I've a couple of issues:
First, ADTPro tell's me "the selected port is in use" I try several times until it says conected.
Second, ADTPro can't see any file. I try to change the work directory whatever folder I choose it's empty (In fact it's not there are files but the application doesn't see them even all type of files selected).
-Windows 10. (Firewall disabled)
-ADTPro 2.0.1
-JRE jre-8u321-windows-x64.
-Serial connection via USB to Serial cable (prolific).
Thanks.

Drakepirate, 2022-02-13:
The ports are well known sometimes 7 9 or 5 depending on the USB port used but the message is the same "port being used".

Drakepirate, 2022-02-16:
Haha, ok here we go...
[image]

Drakepirate, 2022-02-16:
[image]

Drakepirate, 2022-02-16:
[image]

schmidtd, 2022-02-16:
So which COM port does the USB dongle add to the system?

Drakepirate, 2022-02-16:
COM 5

david-schmidt, 2022-02-16:
Ok, COM5 is actually getting a "port busy" exception.  This is going to end up being a fight with the driver and the device.  The best way to go about testing is this sequence of events:

1. Unplug the USB adapter
2. Reboot the computer
3. Start up ADTPro server (with the USB adapter still unplugged)
4. Note which COM ports are available to ADTPro
5. Plug in the USB adapter
6. Note which COM port appears as new (we expect COM5), select whatever is the new one
7. Push the serial button
8. Note what exception you get, if any

If you get the "busy" exception in this scenario, then there's a problem with the serial device driver.

Drakepirate, 2022-02-16:
When I do that No com ports are available, then I connect the cable it  depends on the usb port I select com5, 7 or 9. I use an arduino connected with that cable used as RS323 serial wifi modem and works fine on my pc.

david-schmidt, 2022-02-16:
You might try going back to v2.0.3, which uses a different serial library - it may not work any more due to Java limitations, but might be worth a try:
https://github.com/ADTPro/adtpro/releases/tag/v_2_0_3

But I still suspect the issue is in the interaction of the driver and the Prolific adapter with the Java library.

Drakepirate, 2022-02-17:
Funny, I changed to v2.1.0 cause v2.0.3 ctd after hit connect.