ADTPro/adtpro #147: 2 New Errors using Yellowstone UDC and CFFA3000

joelmcw, 2022-03-24:
David, I have a couple of new issues I've found using Steve Chaimbelin's new Yellowstone UDC with ADTPro.

I have an Apple //e Platinum with the following configuation: 
Slot 7 - CFFA3000
Slot 6 - Yellowstone UDC
Slot 5 - Virtual floppy drive (CFFA)
Slot 4 - Mockingboard
Slot 3 - VidHD
Slot 2 - Super Serial
Slot 1 - Fastchip //e
Slot 0 (Aux) - GW8RAM (emulating Ramworks III)

I currently have 2 HD volumes on the CFFA CF card which get mapped to slot 7,1 & 7,2.  I have one physical 5.25 drive which is mapped as 6,1 and I have 2 Unidisk 3.5 drives attached to the second port on the Yellowstone UDC so they are mapped as 6,2 and 4,1 by ProDOS.  My CFFA virtual Disk II are mapped as 5,1 & 5,2. 

Running ADTPro 2.10 I've found 2 issues.  If I attempt to format the 3.5 disk in 4,1 I get "NO UNIT IN THAT SLOT AND DRIVE". Keep in mind it is the second 3.5" Unidisk that is daisychained to the one that is mapped as 6,2. However, when the drives are scanned it can see the disk and the volume name.  Also, in ProDOS I can see the image and execute programs, catalog, etc.  This might be related and I need to reach out to John Brooks/Peter Ferrie but Bitsy Boot crashes to the monitor when I try to boot 4,1 as well.

The 2nd error message comes in when I attempt to receive/write an image to the same device.  I get the "IMAGE/DRIVE SIZE MISMATCH!" error however if I tell it to go ahead and copy (with a disk that I have preformatted) it receives and writes the 800k image just fine.

david-schmidt, 2022-03-24:
Thanks for the report.  Given what I know about the internals of John's ProDOS and the way we poke around in the device table, there's plenty of wiggle room for things like that to go wrong.  It would be quite helpful of a UDC would arrive at my doorstep to help diagnose.  But in the mean time, what happens when you run with a (vastly) simpler setup, such as just the UDC and SSC?

joelmcw, 2022-03-26:
Same exact results with enhanced Apple //e, 80 col RAM card, Super Serial, and Yellowstone disk controller. 5.25 drive on primary connector and mapped to S6,D1.  3.5 Unidisk (with a second 3.5 Unidisk daisychained) attached to secondary connector mapped to S6,D2 and S4,D1 respectively.  ADTPro doesn't see disk in S4,D1 when trying to format but will write an image to it.  ProDOS has issues with it and won't boot S4,D1 either so I suspect the issue is in ProDOS and just passing along the problem to ADTPro.

joelmcw, 2022-03-29:
Hello David, I'm definitely game to help test.  This is an unenhanced //e with only 80 column card and Yellowstone with drives attached as follows: Unidisk 5.25 on drive 1 connector, Unidisk 3.5 on drive 2 connector with another Unidisk 3.5 daisychained behind it. 

See the ADTPro format screen and then the results of the ProDOS Online MLI call below.
 
[image]
[image]

david-schmidt, 2022-03-29:
Fantastic, thanks for that.  

One more test, if I may: boot up ProDOS 2.4.2, and just get to the BASIC prompt.  I'd like to see a dump of a particular memory area in ProDOS:

1. CALL -151
2. BF31.BF3F

joelmcw, 2022-03-29:
Here ya go.

[image]

david-schmidt, 2022-03-29:
That's interesting.  The 4/1 device is in the device table in ProDOS, but it's not sent in the ONLINE call.  (That may be due to the exerciser boot disk containing ProDOS 1.0.1.  Might be interesting to boot 2.4.2 and run the exerciser from there to see if the reply from ONLINE is different.

joelmcw, 2022-03-29:
Yeah, that seems to explain it.

[image]
[image]

joelmcw, 2022-03-29:
By the way, I reported what I thought was a related issue with BitsyBoot not being able to boot S4,D1 to John Brooks.  It was my misunderstanding that all Sx,D1 drives would be bootable but apparently not.  You likely already know this but here's a link to my report and John's reponse. 
 https://github.com/ProDOS-8/ProDOS8-Testing/issues/63

david-schmidt, 2022-03-29:
Ok, awesome.  That's all I need - I appreciate the legwork.  The device table and ONLINE results tell the tale.

apple2geek, 2022-03-29:
ProDOS versions prior to 1.2 will ONLY recognize and deal with two attached devices per slot.  Period, full stop.
ProDOS versions starting with 1.2 but prior to 2.x will "re-map" ("mirror" is Apple's proper term for it) a SmartPort controller in SLOT 5 ONLY if it has more than two devices such that the extra devices will appear as Slot 2 devices.
ProDOS 2.x and later will map ANY SmartPort controller in ANY slot if it has more than two devices such that the extra devices will appear in any unused slot/drive spots including S3/D1.
Only the first physical drive PER CONTROLLER is bootable.  Mapping ("mirroring") is done by ProDOS only after it has booted up, and has no effect on which devices are bootable.

joelmcw, 2022-03-29:
Apple2geek, I understand that now as John provided those details. Thanks.

David, please let me know if you need me to do any further testing.

david-schmidt, 2022-03-30:
Joel -

I forgot to ask for the driver address table which is in the same neighborhood.  So next time you're in there:

1.    CALL -151
2.    BF10.BF3F

joelmcw, 2022-03-30:
Here it is. Just let me know if you need anything else.

[image]

david-schmidt, 2022-03-30:
Perfect.  Nothing more, I'm collecting samples from multiple users in various scenarios to get good device coverage.

joelmcw, 2022-03-30:
Ok, great. Thanks for all of your support to keep ADTPro alive and well.