ADTPro/adtpro #150: Speediboot With Multiple SSC

michellybells, 2022-04-04:
My Apple IIe has two SSCs in Slot 1 and Slot 2. I connect to ADTPro using Slot 2, however when Speediboot reaches the MLI section and searches for SSC, it finds the one is Slot 1 first and defaults to it. Is there any way to make it detect if multiple SSC cards are present and prompt which slot to use if there's more than one? Otherwise I have to use the much slower old bootstrap method or change my A2pi setup entirely. Thanks!

michellybells, 2022-04-04:
That's a good point. I originally had an issue where booting with Speediboot would connect properly over serial, but booting with the boot disk would not connect. However, I just tested it and doing it with the boot disk is working fine, so I'll just do that from now on.

david-schmidt, 2022-04-04:
It’s a valid concern, as the automatic slot scanning code counts up from slot 1 and stops when it finds a viable communications card.  It’s rare but possible that someone has two, and this is the problem that happens in that situation.

In daily ADTPro operations, you can save a configuration file that codifies the slot, so that’s how you get around it in general. But speediboot needs to make some kind of determination, and I haven’t come up with a way to specify to skip slots in the scan.

closed 2022-04-04 by david-schmidt