ADTPro/adtpro #151: Installing Up To Date JDK/JRE using Mac Homebrew Breaks ADTPro

kaplanfx, 2022-04-17:
I was scratching my head on this one for a while, I had a working ADTPro setup that seeming suddenly stopped working. After you install OpenJDK using Mac Homebrew (brew command) ADTPro will use the updated JRE. It launches and seemingly works, however the change working directory command is broken (it won't list the comments of certain directories) and when you request to receive a file over serial (to a IIgs in my case) you will see the request appear on the ADTPro host on the mac indicating that the devices are talking, however the IIgs always displays a Host Timeout error. Uninstalling the OpenJDK fixed the problem, however now the other application that requires it doesn't work.

Is there a way to use a specific JRE with ADTPro? Trying to run the jar directly from the command line didn't work.

david-schmidt, 2022-04-18:
I'll have to look into why this particular JDK distro is misbehaving.  But in the mean time, yes, for sure, you can decide what JDK to use.  Apple tries to simplify things for you, but sometimes it just ends up making it more complex. There is a shell script that comes as part of ADTPro called "adtpro.sh" that everyone else on a *ix platform uses that explicitly defines where to source the Java runtime from.  You can tailor that to your environment and use that rather than just double-clicking on the ADTPro icon.

kaplanfx, 2022-04-18:
This worked, although it took me quite a while to figure out that Apple had the java binaries placed under: /Library/Java/JavaVirtualMachines

The shell script appears to be called ADTPro.command on MacOS though, rather than adtpro.sh