ADTPro/adtpro #153: Starting functional prodos over the casette port

jordancrubin, 2022-05-26:
Listed as one of the given scenarios below does not include the possibility for prodos running without adtpro.

Choose a bootstrapping scenario below that matches your situation.
    [If you have no software now and only want to start DOS 3.3 over the cassette port]

It look from the instructions that pulling prodos from the audio is just one step to get adt started but is not a free standing raw image with a basic.system or utilities to format disks or otherwise.   Is it possible to add such an option, to bootstrap a prodos utility disk onto the //e

thanks,

j

schmidtd, 2022-05-26:
What you're looking for doesn't actually work that way in ProDOS.  If, for example, you boot a ProDOS floppy that has the `PRODOS` and `BASIC.SYSTEM` files on it, you'd be able to boot to a BASIC prompt and be able to `LOAD` and `SAVE` BASIC files, but that would be about it - you can't format a floppy without the ProDOS utilities program (or ADTPro because it also happens to contain a formatter).  DOS 3.3, on the other hand, has `INIT` built into it and so it can format a floppy with just the bare DOS and nothing else.  That's why the bootstrapping scenarios talk about getting ADTPro up and running so that you can then send other full disks across the connection (such as ADTPro itself) to get you further down that path.

david-schmidt, 2022-05-26:
What you're looking for doesn't actually work that way in ProDOS.  If, for example, you boot a ProDOS floppy that has the `PRODOS` and `BASIC.SYSTEM` files on it, you'd be able to boot to a BASIC prompt and be able to `LOAD` and `SAVE` BASIC files, but that would be about it - you can't format a floppy without the ProDOS utilities program (or ADTPro because it also happens to contain a formatter).  DOS 3.3, on the other hand, has `INIT` built into it and so it can format a floppy with just the bare DOS and nothing else.  That's why the bootstrapping scenarios talk about getting ADTPro up and running so that you can then send other full disks across the connection (such as ADTPro itself) to get you further down that path.

jordancrubin, 2022-05-27:
Thanks for the swift reply dave, and a wonderful programme.  I guess short
of an ability to bootstrap an entire prodos utility disk, which would be
cool, this would be the extent of what could be done through the audio
port. Dos 3 being the exception

david-schmidt, 2022-05-27:
That's the thing - bootstrappping over serial doesn't send a _disk_, it sends a _program_.  Sometimes that program is ProDOS itself; sometimes it's DOS; sometimes it's ADTPro.  But in no case does it send an entire disk.