ADTPro/adtpro #154: Cannot reveive when I boot on the disk on Apple IIc

Neo2003, 2022-05-30:
I have this problem. I use the version 2.1.0.
I am able to use ADTPro to launch the software from my PC portable with a serial cable I made
I send prodos, then adtpro and no problem, I can receive .dsk and write new disks.

But if I boot from the adtpro 2.1.0 disk I made, I cannot receive anything, I have a timout each time, so I have to bootstrap each time I want to make a new disk, I cannot use the Adtpro disk at all.

david-schmidt, 2022-05-31:
That's an interesting set of circumstances.  When you're doing the initial bootstrapping, are you able to use Speediboot?   You mention sending the two pieces separately, but that shouldn't be necessary... Speediboot should be able to get you up and running all by itself.  If it can't, that's an interesting additional data point.

Neo2003, 2022-05-31:
I never tried Speediboot, I will do later today and report back, thanks

david-schmidt, 2022-05-31:
There's really nothing different in the comms engine between the bootstrapped version and the version on disk.  It's likely down to something environmental, and given nothing else to go on, I'd suspect the USB adapter (assuming you're using one of those).

What happens if instead of asking for a directory of files you ask to send a disk image back over to the host?  So boot the ADTPro disk and then go right into Send?