ADTPro/adtpro #156: Loading problems on MacOS 11.6.7

robbo007, 2022-07-08:
The binary does not open after accepting the security warnings and allow the app to run, when trying to run ADTPro 2.1.0 under MacOS 11.6.7. Clicking does nothing. Does not run.

robbo007, 2022-07-08:
The problem seems to be you have to manually add FULL disk access to ADTPro under System Preferences, Security and Privacy, Privacy. Just dragging it into the Applications folder does not do this nor does it prompt you to add this.

christo, 2023-03-08:
I've noticed ADTPro writes its own configuration file in the installation directory. On MacOS I guess this would ideally live in some other directory for user preferences which wouldn't require such onerous permissions to be granted.