ADTPro/adtpro #157: Communication with Apple IIc hangs on files request - times out. V2.1.0

TheRealAnubiss, 2022-07-25:
Old version I had installed forever (2.0.1) works.  Updated to newest version 2.1.0.  Wrote 2.1.0 boot floppy using 2.0.1.

Boot 2.1.0 on Apple IIc, start ADTPro 2.1.0 on Win7 computer with Java build 1.8.0_341-b10.

Serial connection shows connected.  Using Apple IIc to (R)ecieve files.  Hit Enter at Filename: prompt.  Request shows up on the Win7 PC, but it times out.  Settings checked - same as the old version.  Using Imagewriter cable, and have that option checked.

david-schmidt, 2022-07-25:
Are there any messages in the command window (the black DOS prompt that is usually hidden but starts with Java)?

TheRealAnubiss, 2022-07-25:
Hi,

The Java window is empty except for the message at the top:
ADTPro Server version 2.1.0

I hit 'Disconnect' and it said there was a transfer in progress - cancel Y/N
I clicked Yes, and then the window reads:
java.lang.InterruptedException: sleep interrupted
at java.lang.Thread.sleep(Native Method)
at org.adtpro.CommsThread.sendPacketWide(Unknown Source)
at org.adtpro.CommsThread.sendDirectoryWide(Unknown Source)
at org.adtpro.CommsThread.dispatchCommand(Unknown Source)
at org.adtpro.CommsThread.commandLoop(Unknown Source)
at org.adtpro.CommsThread.run(Unknown Source)

david-schmidt, 2022-07-26:
So it sounds like jostling the adapter's speed is helping it get started.  What kind of serial adapter do you have?

TheRealAnubiss, 2022-07-26:
Hi,

I have an Imagewriter cable with a db25 to db9 adapter.  This works without fault on the 2.0.2 version, which is what I've been using at this point.  I'm using an older Win7 PC with a 9 pin serial port on it.

TheRealAnubiss, 2022-08-02:
I've tried this on a different IIc just to check, and I have the same problem.  My IIc+ with homemade cable has no problem with the newest version.  Same Windows PC for all tests..

david-schmidt, 2022-08-02:
Ok, so the common point of failure seems to be the native serial port on the PC?

TheRealAnubiss, 2022-08-02:
Hmmm - no?  It works fine with the older version ADTPro with no toggling of speeds, etc, and with the IIc+ it works fine with the old and new ADTPro (same serial port) - something seems to be up with the IIc and the Imagewriter cable.

Both Apple IIc's work with the Imagewriter cable and ADTPro version 2.0.2
Neither of the IIc's work with the Imagewriter cable and ADTPro version 2.1.0
The Apple IIc+ works with home made cable and ADTPro version 2.0.2 and 2.1.0

All tests on the same Win 7 PC.

david-schmidt, 2022-08-02:
Ah, so the common failing components are the real Imagewriter cable and the new serial library in 2.1.0.  And the symptom of having to toggle the speed makes this otherwise failing combination work, correct?

TheRealAnubiss, 2022-08-02:
Yes.  The same cable works fine on the older version.  I didn't have a problem until I got the latest release of ADTPro.

david-schmidt, 2022-08-02:
Do you have a USB RS232 adapter you can test in that same windows box?

TheRealAnubiss, 2022-08-03:
I think I have an old Belkin one handy.  I may have to find drivers if Win7 doesn't like it.
I'll let you know for sure tomorrow.  Thanks!

TheRealAnubiss, 2022-08-03:
Hi,

Ok, I luckily have the Belkin F5U109, which I can get the Win7 64 bit drivers for.  It's configured for COM3 by default.

david-schmidt, 2022-08-17:
Oooh, back now - so, what changed between then and now is the entire underlying hardware communications layer.  The old one (rxtx) seemed to require that flag, but tickling the same bits in the new library (jssc) seems to not require it.  So I'd say keep on keepin' on with the box unchecked for now...

closed 2022-08-17 by david-schmidt

TheRealAnubiss, 2022-08-17:
Thank you for your time, and for such a useful program!