ADTPro/adtpro #158: What serial cables do I need to use to connect a MacBook Air M1 with USB-C ports to an Apple IIc serial port?

andymilleville, 2022-07-25:
Title says it all.

I have a MacBook Air M1 running Monterey 12.3.1. I want to use ADT Pro to transfer disk images to my Apple IIc. I have the cable for the Apple IIc and the boot disk from David (thanks David!) as well. However, when I launch ADTPro on my Mac, it can't see any serial ports. It says it needs rxtx installed but that project seems defunct. 

What are my options here? I have a Windows 8 PC but I would like a MacOS solution as that laptop might not make it too much longer.

david-schmidt, 2022-07-26:
The messages still say rxtx, but last release I swapped out jssc as the native serial library.  It looks like jssc added support for M1 a release ago, so if you are adventurous, you can download jssc-2.9.4.jar from https://github.com/java-native/jssc/releases and refer to that jar instead of lib/jssc/jssc-2.9.2.jar in your invocation script (ADTPro.command).  If you just double-click the icon to start it up, though, you'll need to wait for me to re-package and re-release.

andymilleville, 2022-07-26:
I am indeed adventurous. Thank you for the advice. I'll let you know what happens.