ADTPro/adtpro #160: Haba Disk drive not recognized by ADTPro.

TheRealAnubiss, 2022-08-18:
Hi - not sure how to contact you directly, but I have a question.  I have an uncommon Apple II drive called a HabaDisk drive.  It's a 3.5" single sided double density that says it is 360K.  You put the 5.25" boot disk in and then it boots the software from the 3.5" HabaDisk drive.  When ADTPro checks the drives, it hangs because it's not expecting this drive.  I'm not a very good coder, but I'd be interested in getting this drive working with ADTPro so I can image the disks - they're not online that I can find, and I'd like to preserve them as much as possible.  What would be some good ideas to try and get ADTPro to recognize this drive?
If this is not the appropriate place for this, please let me know and I'll contact you a different way.
Thank you!
[image]
[image]

david-schmidt, 2022-08-18:
Here is fine to reach me.  So the thing is - ADTPro expects to be able to reach a disk drive by making ProDOS calls.  If this disk doesn't work under ProDOS (i.e. if it doesn't have a ProDOS driver or use a ProDOS-compatible interface card) then it's simply not going to work.

TheRealAnubiss, 2022-08-18:
Hi,

I think that's what it uses.  The boot disk says "ProDos" on it, at least, and the main program on the 3.5" looks like Appleworks with some modifications.  It has folders on the disk, etc.  Does that sound right?  I made an image of the boot disk that I could send you if that would help.  ADTPro accesses the drive, but it just keeps running as I'm assuming it's not getting back the answer it's looking for size wise.  320K isn't what either the typical 5.25" or 3.5" drive has.

david-schmidt, 2022-08-19:
Ok, so it looks like it's using ProDOS of some stripe.  You can just drop the image of the boot disk here in a comment...

TheRealAnubiss, 2022-08-19:
Hi,

It plugs right into the back of the IIc's drive port.  I think with the IIe it needs the newer drive port, not the card with pins and ribbon cables that run out the back.

I'll pop in the 5.25" boot disk image - basically, it boots up and says HABADISK, then looks for the software bundle disk in the 3.5" drive.  If it doesn't see that then it gives a 'not found' error.
[HabaDisk_boot_prodos_1991.zip](https://github.com/ADTPro/adtpro/files/9379086/HabaDisk_boot_prodos_1991.zip)

david-schmidt, 2022-08-21:
Ok, thanks for posting that.  They did hack a version of ProDOS and overwrote Apple's copyright messages.  I verified that ADTPro is happy running under their hacked ProDOS with normal disks, but of course I don't have a Haba to test with.  What we probably need to do next is fully boot with the 3.5" disk and see what Haba writes to the devices table.  If you can boot all the way up and do the dumps (there were several) in the chain of this issue:
https://github.com/ADTPro/adtpro/issues/147#issuecomment-1082009186

That should give me what I need to see what ProDOS thinks of this disk.

TheRealAnubiss, 2022-08-21:
Ok, so I'll write out the prodosexerciser.dsk, and boot that disk with the habadisk 3.5" connected with the software bundle disk in it, follow the directions and screenshot the info, then download and write out the prodos2.4.2.dsk, boot that, get to the basic prompt and screenshot the results of:
CALL -151
BF10.BF3F
Does this sound correct?

david-schmidt, 2022-08-21:
The fact that it's choking on the $C5 call is not a good sign for the drive itself.  ADTPro also makes this call at various times. 
 What slot/drive does it appear as in ProDOS if you're using it normally?

TheRealAnubiss, 2022-08-21:
That's the thing, actually - there's no using it normally (I'm not super familiar with ProDOS, but I can get up to speed if needed).  The only way I've gotten it to work is to use the 5.25" boot disk with the program bundle disk in the 3.5" habadisk drive.  The only thing that runs on the 3.5" drive that I can find is the program bundle and the templates disk (which are in the picture of the software).
I can boot up that ProDOS disk I just made (2.4.2) - then what should I do?

david-schmidt, 2022-08-21:
[image]
The box says it will tell you how to copy AppleWorks to a 3-1/2" disk - that would go a long way to telling us how they expected us to interact with this drive in a ProDOS world.  Do you have any of the documentation that talks about that?

TheRealAnubiss, 2022-08-21:
I'll get on that - I only have the HabaMerge book, but it covers a lot of stuff.. I'll get on that!  Thanks!

schmidtd, 2022-08-22:
So, that doesn't bode well if they have some proprietary thing that does copying actions.   As best I can tell from the output of the $C5 ONLINE call, it might be slot 4 drive 1.  So if you boot your system with their booters, can you get to the BASIC prompt?  I'd be interested in what `CAT,S4,D1` looks like.

schmidtd, 2022-08-22:
...but this might be a tempest in a teapot, as it doesn't seem like this is a normally-supported ProDOS device - meaning ADTPro won't be able to manipulate it.

david-schmidt, 2022-08-22:
So, that doesn't bode well if they have some proprietary thing that does copying actions. As best I can tell from the output of the $C5 ONLINE call, it might be slot 4 drive 1. So if you boot your system with their booters, can you get to the BASIC prompt? I'd be interested in what CAT,S4,D1 looks like.

...but this might be a tempest in a teapot, as it doesn't seem like this is a normally-supported ProDOS device - meaning ADTPro won't be able to manipulate it.

TheRealAnubiss, 2022-08-22:
That's something I haven't tried.  I believe I can get it to the basic prompt, but I'll have to see for sure.  I'll let you know, and thanks for the help so far!