ADTPro/adtpro #164: Configuration from Headerless mode

sairuk, 2022-11-21:
We have added this tool to RetroNAS and were looking for some guidance on passing configuration options to the headerless launch, either through env vars, env file or cli something to that effect.

Is this possible/planned? 

Thanks

Refer: https://github.com/danmons/retronas/commit/d116559aca5fe97b2fc75e0372370ecef29fd143

david-schmidt, 2022-11-22:
The way external adopters have approached this in the past is to script things such as start and stop (start is easy enough; stop is some combination of `ps` and `kill -9`).  You have full control of all configuration via the config file `ADTPro.properties` - if you shut ADTPro down, modify that file, then start it back up, it'll use those configuration values.

The full raft of things it writes isn't really documented, but some examples are:

```
#ADTPro.properties
#Tue Nov 22 09:25:11 EST 2022
CommPort=/dev/serial
CommPortSpeed=115200
CommPortBootstrapSpeed=9600
WorkingDirectory=/some/where
SerialIPHost=localhost
CommPort=COM1
CoordX=373
CoordY=3
CoordH=222
CoordW=311
CommPortBootstrapPacing=250
HardwareHandshaking=false
TraceEnabled=false
SerialIPPort=1977
```

And then you can have it start up as connected in whatever mode you want by invoking it with e.g. `adtpro.sh headless serial`.

sairuk, 2022-11-22:
thank you for this information, exactly what i was looking for.

I was not seeing a config generated on headless run, so was curious how to go about this.

closed 2022-11-22 by sairuk