ADTPro/adtpro #165: "Unable to open file" error in ADTPro batch mode

ApplePerson1, 2022-12-28:
After successfully running the bootstrapper over a serial cable connection between a PC and an Apple IIe, and selecting the "B" option, I can enter a file name, "UPLOAD."  On pressing return, I get the "select source volume" menu where I then select slot 6, drive 1, my floppy disk drive containing a PRODOS formatted diskette with the target programs on it.  At this point however, I immediately get the "Unable to open file" error message.  The ADTPro program doesn't even try to access the drive.

What am I doing wrong?

david-schmidt, 2022-12-28:
Take a look at the command/terminal window "behind" the ADTPro client - do you see an error message coming from there?  It's talking to the host and having trouble finding the location you're asking to save files to for one reason or another.

If you just have one disk to send, you wouldn't want to use the Batch function - you'd just use Send.

ApplePerson1, 2022-12-29:
Thank you for the fast response.   I did try using "Send" instead of "Batch," but am getting the same error message. 

 I'm afraid that I do not understand some of the terminology.  When the program says that it is "unable to open file," is it referring to a file on the PC--i.e., the destination file--or the file on the Apple II disk, the source file, that I am trying to send to the PC?  In any case, this is what happens: the window on the PC end starts off reading "Request: image from Apple to host."  On the Apple end, I then select "S" for send, it asks for a "filename," and I enter "TTT1.DSK."  (This is a new, made-up file name.  I presume that this is what it wants, because when I tried entering the name of a file that already exists on the PC, it tells me that that file already exists, and asks if I want to copy over it anyway).  At the moment I hit ENTER on the Apple, the window on the PC changes to the message "Request: query file size," and below that, "TTT1.DSK."  I then, on the Apple, am presented with the "select source volume" menu, where I select "6  1  BLANK  280," at which point I immediately get the "unable to open file" message, and the window on the PC changes back to "Request: image from Apple to host."  

Any further suggestions greatly appreciated.  Thank you.

ApplePerson1, 2022-12-29:
Problem is solved, transfer successful.  On further considering your comment, it became clear to me that the "unable to open file" message meant that the Apple II was unable to create the destination file, TTT1.dsk at the PC end.  I then tried to create it manually using Windows (10) file explorer, and failed, because the entire ADTPro folder on the PC was write protected.  It was write protected because, when I downloaded it, I unzipped it into the C:\ProgramFiles folder, where Windows 10 write protects everything.  So the solution was to move the ADTPro folder (with all its subfolders) to the main C:\ folder, which removed the write protection, and thus allowed the Apple II to create the target TTT1.dsk file in the Disks subfolder.

david-schmidt, 2022-12-30:
Sounds great - glad you got that figured out!

closed 2022-12-30 by david-schmidt