ADTPro/adtpro #166: Apple II GS null-modem serial cable (custom build) do not work properly

Dwlads, 2023-03-08:
Hi Guys,

First of all, Kudos for this fantastic project :)

I made a custom null-modem serial cable for my Apple II GS and looks like I did something wrong. I Followed the cable diagram available at adt.com/connectionsserial.html, as follows:

[image]

So my cable connections was made like this:

[image]

My question is simple: Did I do something wrong? 

My solder skills are good, and I checked cable continuity, and everything looks fine, but I received a "Host Timeout" message when I tried to perform a (D)DIR on Adtpro 2.1.0.

Thank you and best regards.

david-schmidt, 2023-03-08:
The short answer is, "it depends."  Assuming the pinout you're describing looks the same and has the same pin orientation as the pictures in https://adtpro.com/connectionsserial.html#MiniDIN8 , then the cable seems like it's reasonably well wired.  But there's a lot more variables in play.  The host computer's serial connection (whether native or USB-based), the COM port it's really using vs. the one you selected, the plug you used in the back of the GS... all are important to get right, and will have the exact same effect with "Host Timeout" if they're wrong.

Dwlads, 2023-03-14:
Hi David,

As mentioned in our e-mail exchange, I finally had a successful cable connection. I used a better-quality Min-Din 8 connector and changed how I soldered the wires. Below is a simple scheme showing how I did this time:

[image]

In addition, I hold the Min-Din 8 connector during the soldering process using Blu Tack to transfer the heat and avoid connector melting. 

So, this is the end of my custom cable saga, and again a BIG THANK YOU for your support and Kudos for this fantastic project.

Best regards.

david-schmidt, 2023-03-14:
Awesome, glad you got that working.  Enjoy!

closed 2023-03-14 by david-schmidt