ADTPro/adtpro #168: 100% CPU after running for some time

hackerb9, 2023-03-14:
Is it a known issue that ADTPro server ever uses up 100% of one CPU core? I had left the server running unattended after doing some transfers and when I came back the next day `top` showed that java was using up 100% CPU. I restarted the program, and the problem has not  reappeared, yet. 

Possibly relevant: I have "Activate Trace" turned on. I am using version ADTPro 2.1.0 with openjdk 18.0.2-ea 2022-07-19.

david-schmidt, 2023-03-14:
Not a currently known problem, no.  I would definitely see if it reproduces without tracing on.

hackerb9, 2023-03-17:
I've been running it for days and I can't reproduce the problem at all now, so I'll close this. Maybe the server had gotten into a funny state after the client crashed when trying to receive a NIB.

closed 2023-03-17 by hackerb9