ADTPro/adtpro #177: ADTPro Apple III Bootstrap stuck at "OK"

MicroCoreLabs, 2023-07-27:
I am using the latest version of ADTPro (on multiple laptops). I first enter the small program on the Apple III then begin process on the server laptop. I see the ADTPro logo and the progress characters changing.  But after the progress bar is complete the banner on the Apple III just says " ///   OK  ///" and there is no more activity.

If I reset the Apple III Wirth open-apple,Control, and RESET and enter A000.A050 the program that I entered is changed to mostly 0x00's and 0xFF's.

I have an earlier Apple III which has the corrugated sheet metal on the bottom of the chassis and also contains a 12V memory card.  So I was wondering if my machine's BIOS is too early to be compatible with ADTPro, or perhaps there is a problem with my machine?    When I power up the machine it tries to boot from the diskette and does not enter the debug memory checking mode.

MicroCoreLabs, 2023-07-27:
I get the message_1 LOADING SOS KERNEL:  but then it seems to jump to "   ///   OK   ///   "    

What is causing the intermediate two steps to be skipped?

message_1:
	.byte	" LOADING SOS KERNEL:"

message_2:
	.byte	"   LOADING ADTPRO  :"

message_3:
	.byte	"LOADING SOS DRIVERS:"

message_4:
	.byte	"   ///   OK   ///   "

MicroCoreLabs, 2023-07-27:
java -version
java version "1.8.0_333"
Java(TM) SE Runtime Environment (build 1.8.0_333-b02)
Java HotSpot(TM) Client VM (build 25.333-b02, mixed mode, sharing)

david-schmidt, 2023-07-27:
It definitely sounds like you've got a hardware problem on your hands.  We had this exact same question yesterday: https://github.com/ADTPro/adtpro/issues/176

MicroCoreLabs, 2023-07-27:
Ok.. I see. looks like a potential HW issue...

I am trying to create a boot disk using ADTPro on an Apple III, but I did not know that I can use my existing Apple II+ setup to generate an Apple III disk.  I thought there was a boot sector incompatibility...  But if I can use ADTPro and my II+ to make the ///  disk then I can try the dealer diagnostics as you mentioned.  Hopefully I understand this correctly.

david-schmidt, 2023-07-27:
You absolutely can mix and match, either way.  Boot disks are particular to a machine, but you can create any with either.  Make the dealer diagnostics disk and also make an ADTPro SOS boot disk while you're at it... you may see a different failure mode if your hardware is at fault.

MicroCoreLabs, 2023-07-29:
I locked the Apple III bus speed to 1 Mhz and swapped out the 32K DRAMS with 4116 16K DRAMs on the 12V memory board and I get a lot further in the bootstrapping process , but it always gets stuck at this point.  Does this indicate a DRAM issue?  Do you know if it is allowed to swap the 32K DRAMS with 4116's?  The schematics says that you can, but maybe SOS doesnt like holes in the memory range(?)

Picture enclosed.
[image]

david-schmidt, 2023-07-29:
Sorry, ADTPro isn't a great diagnostics tool for the /// - you'll be much better off creating a diagnostic disk (with a II) and running that to get the hardware sorted.  SOS and ADTPro use the whole machine, so it really all has to be in good running order before it'll work.

MicroCoreLabs, 2023-07-30:
I got my machine to work and was successful getting ADTPro to create some bootable disks!  Yay!

Can I make a suggestion for your next version?   Could you add a small memory test as part of the bootloader to let the user know if memory is usable so that ADTPro will work?  I know that it is not a diagnosis tool, however I have experienced that a lot of time can be spent typing in the boot loader assembly and watch lots of downloading progress succeed, only to find that it locks ups somewhere in SOS because ADTPro was writing to memory which had bit errors.  Thankfully someone else provided me a diagnostic diskette to trace down the bad DRAMs, but the point of ADTPro is to allow the user to be self-sufficient and not need loaner diskettes. :).   A small memory test would save a lot of time by letting the user know ahead of time that there are DRAM errors so that ADTPro will not just forge ahead and download to bad memory.

david-schmidt, 2023-07-31:
Well, that's an interesting idea.  There is something similar for audio in that there's a test regimen to help tune audio levels:
https://adtpro.com/connectionsaudio.html#Testing

It would be a bit bigger than a breadbox since it would have to deal with bank switching (we'd be running in one of the swapped-in banks at the time this would be active), but would be fun to contemplate.

closed 2023-07-31 by david-schmidt