ADTPro/adtpro #178: Legal (licensing) question

colinleroy, 2023-09-07:
Hello folks,
And thanks for ADTPro :)

I have a quick question - I want to publish a pull request to the cc65 project (https://github.com/cc65/cc65) to prevent its SSC serial driver from "succeeding" opening a serial port using its ACIA 6551 driver on a IIgs. The cc65 ACIA driver checks for card identification bytes before succeeding, but as we know (http://www.1000bit.it/support/manuali/apple/technotes/misc/tn.misc.08.html) they're the same bytes on an SSC card, on //c, and on IIgs.
Afterwards, havoc ensues :)

Looking for information on how to differentiate the 6551 and the Z8530 based cards, I found a good way to do so in https://github.com/ADTPro/adtpro/blob/f412f07a801295d4f70a582da9be0aa14e1b75cb/src/client/dos/adt.s#L700

Now, ADTPro is GPL2 licensed where cc65 is zlib-licensed, and the cc65 ACIA driver is public domain.

Do you consider these 6 lines of code to be derived from ADTPro or do you consider that using this offset/byte combination I saw in adt.s to not be copyrighted ?
```

+IIgsRomOffset := $1B
+IIgsRomInstr  := $EB
[...]
+        ; Check the ROM to verify this is not a IIgs
+        ldy     #IIgsRomOffset
+        lda     #IIgsRomInstr
+        cmp     (ptr2),y
+        beq     NoDevice

```
Thanks in advance!
Colin

david-schmidt, 2023-09-07:
Hi, Colin - glad you like it. :-)

So, I wrote that code in ADTPro and backported it to ADT.  They're both covered by GPL2.  I'm not sure if you're asking if the technique for discerning a GS is a concept you'd consider copyright(able) or if lifting this few bytes would be considered a derivative work, but it seems to me it's both of those things.

IANAL, but if I lift a concept from someone else, which I've done a lot, I always give them credit (i.e. see https://adtpro.com/credits.html) as well as keep copyright statements inline (i.e. see https://github.com/ADTPro/adtpro/blob/main/src/client/prodos/format.asm#L19).  As for merging incompatible licenses, I'd again have to stress that IANAL and am not really qualified to comment on that.  My code remains covered by GPL2 and you're free to use it in any way compatible with it. :-)

colinleroy, 2023-09-07:
Hi David,
Thanks for your quick response!
I'll find another way for the cc65 driver then :)

Colin

closed 2023-09-07 by colinleroy

colinleroy, 2023-09-07:
I'll simply go with the byte at Cn00, seems unique enough too 🙂