ADTPro/adtpro #184: Mac OS 14?

TheArt01, 2023-12-10:
Hi, I have a problem starting the ADTpro server on my Macbook. I can start the application, but as soon as I click on "serial", I get an error message: "no serial ports or no serial library"
[image]

What am I doing wrong?

david-schmidt, 2023-12-10:
Do you have a USB-RS232 device plugged in?  What happens if you click on File->Serial Configuration?  Can you pick the serial device you have plugged in?  (Also... what version are you using?)

TheArt01, 2023-12-11:
Hi David,

I’ve tried 2.0.3 und 2.1.0.

Yes, I have plugged in the correct Serial Cable into the //c (worked previously with another ADTpro server), attached a Serial to USB A and USB A to C adapter.

Under 2.1.0. I cannot click on File->Serial Cond.
Under 2.0.3. it shows the attached.

Latest Java and OS X is installed.

Thanks for helping me

Carsten

￼Do you have a USB-RS232 device plugged in? What happens if you click on File->Serial Configuration? Can you pick the serial device you have plugged in? (Also... what version are you using?)

david-schmidt, 2023-12-11:
2.0.3 absolutely won't work, and that's why 2.1.0 exists - to get Java up to date, especially for Mac, which likes to shift the sand out from under you.  If you "cannot click on File->Serial Configuration" that is a problem - can you explain more what that means?

TheArt01, 2023-12-11:
Hi David, 

yes sure, I’ll try to.

If I click on the „Serial“ button (the one with the cable), I get the following error message: „There are no serial ports, or the serial library RXTX is not installed or visible"

If I click on File -> Serial Configuration, nothing happens under 2.1.0.

Under 2.0.3., I get the Port/Bootstapping box with Port set to „..“ (unable to change that), and a speed of 115200, which I can change to 19200.

In both versions, the server’s main windows shows „No Activity“ and „Not Connected“

On the //c’s side, everything should be fine. I have used the computer, the null-modem and the Rs232 to USBA-adapter in the exact-same configuration on my old Intel iMac. Now I am trying to connect everything to an M1 Macbook, using an additional USBA to C adapter by apple.

I really appreciate your help. Thank you, David!

BR
Carsten

david-schmidt, 2023-12-11:
What this all tells me is that either your Mac isn't seeing your serial port, or there are some permissions problems to sort out.    So let's start with the system - if you go to your System Information 
(Mac icon -> About this Mac -> (scroll down to) System Report...) do you see a USB device that sounds like an RS232 port?
[image]

TheArt01, 2023-12-11:
I can see a „USB Serial Controller D“

> 
> 
> What this all tells me is that either your Mac isn't seeing your serial port, or there are some permissions problems to sort out. So let's start with the system - if you go to your System Information
> (Mac icon -> About this Mac -> (scroll down to) System Report...) do you see a USB device that sounds like an RS232 port?
> Screenshot.2023-12-11.at.11.55.24.AM.png (view on web) <https://github.com/ADTPro/adtpro/assets/1110325/c1aba887-a708-4afa-873e-ea16b712f66d>
￼

quickthyme, 2023-12-21:
Just wanted to chime in that I am experiencing this exact same issue. Using the same RS232-USBA cable that I used with my old Intel MacBook Air, with Apple branded USBC-A dongle, now on M1 MBP, getting same error when trying to enable Serial, and not able to bring up Serial Configuration from the menu at all just as @TheArt01 described. I am on macOS Sonoma 14.2.1 with JRE 18.19 for arm64 from Oracle. System profile report does show the device as being recognized `FT232R USB UART`.

david-schmidt, 2023-12-21:
I did a test build with an updated serial library, and I think that'll solve the problem.  I got called out of town so my build facilities aren't available to me just now, but I think that'll do it.

quickthyme, 2023-12-21:
Well, that's good news then! Thanks for looking into it. Does this mean we'll be seeing a "2.1.1"? or will it be more like a "2.1.0 Special Edition™"?

david-schmidt, 2023-12-21:
We'll do a testcase build, and if you validate it (I don't have fancy Apple M1/2 hardware to test with) we'll release a new version generally.

TheArt01, 2023-12-22:
Hi David,happy to validate the testcase build whenever you are ready.BRCarstenSent from my iPad. Please excuse typos and/or abbreviations.

IdreesInc, 2024-03-14:
I am also encountering this issue, has anyone determined a resolution?

david-schmidt, 2024-03-14:
Here, give this a try:
https://github.com/ADTPro/adtpro/releases/tag/applesilicon

IdreesInc, 2024-03-14:
Thank you, I believe that is working as expected! I say believe because the app now allows me to attempt to bootstrap and sends commands over to the Apple IIc, but I am receiving garbled syntax similar to on my Windows machine. Seems like that is likely unrelated to this issue but including the below photo just in case:
[image]

david-schmidt, 2024-03-14:
> Thank you, I believe that is working as expected! I say believe because the app now allows me to attempt to bootstrap and sends commands over to the Apple IIc, but I am receiving garbled syntax similar to on my Windows machine.

Yep, nope, this is a speed mismatch problem.  At least now the bits are getting on to the wire.